betaflight/betaflight-configurator #3919: Reciever Tab model preview not responding.

E120103016, 2024-04-29:
### Describe the bug

Even though the channel bars respond correctly to stick/switch inputs, the preview of the model does not move at all.Betaflight OMNIBUS F4 V3 . reverting to 10.9 shows model preview move correctly.

### To Reproduce

Power radio. Connect to config 10.10.0 Go to receiver tab, move sticks - observe bars move correctly, but not model preview at all.

### Expected behavior

model preview to move in accordance with stick inputs/channel bars

### Configurator version

10.10.0

### Flight controller configuration

```txt
No response
```

### Add any other context about the problem that you think might be relevant here

No response

E120103016, 2024-04-29:
1.Betaflight OMNIBUS F4 V3 . reverting to 10.9 shows model preview move correctly.(SUBS)
2.Power radio. Connect to config 10.10.0 Go to receiver tab(SBUS), move sticks - observe bars move correctly, but not model preview at all.

E120103016, 2024-04-29:
Betaflight Firmware 4.5.0.Power radio. Connect to config 10.10.0 Go to receiver tab(SBUS), move sticks - observe bars move correctly, but not model preview at all.

nerdCopter, 2024-04-29:
duplicate of https://github.com/betaflight/betaflight-configurator/issues/3853 and https://github.com/betaflight/betaflight-configurator/issues/3868. seemingly fixed in https://github.com/betaflight/betaflight-configurator/pull/3847 but maybe render-modal is still disconnecting.

haslinghuis, 2024-04-29:
Duplicate of #3868

Was not able to reproduce.

closed 2024-05-05 by haslinghuis