betaflight/betaflight-configurator #3920: Telemetry slider appears twice

hydra, 2024-04-29:
### Describe the bug

[image]

### To Reproduce

Flash SPRacingH7RF binary, view receiver tab.

The exact binary used for testing was the one from the SPRacingH7RF with PixelOSDSupport from here: https://github.com/betaflight/betaflight/pull/13603#issuecomment-2082342485

### Expected behavior

Only one slider for telemetry.

### Configurator version

`master` - `cd25abfa033e2979102f57e601e0a2d109bc0980`

### Flight controller configuration

```txt
# version
# Betaflight / STM32H730 (SP7R) 4.5.0 Apr 29 2024 / 12:01:30 (17c4fe0c6a) MSP API: 1.46
# config rev: 53c4a3c

# start the command batch
batch start

board_name SPRACINGH7RF
manufacturer_id SPRO

# aux
aux 0 40 0 1300 1700 0 0

# master
set osd_vbat_pos = 2443
set osd_rssi_pos = 353
set osd_link_quality_pos = 2369
set osd_rssi_dbm_pos = 2401
set osd_tim_1_pos = 2433
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2411
set osd_craft_name_pos = 2121
set osd_altitude_pos = 2421
set osd_core_temp_pos = 2456
set debug_mode = RX_EXPRESSLRS_SPI
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 1
set expresslrs_uid = 0,0,68,83,145,153

profile 0

rateprofile 0

# end the command batch
batch end
```

hydra, 2024-04-29:
Also happens using offical build:
```
# version
# Betaflight / STM32H730 (SP7R) 4.5.0 Apr 28 2024 / 21:10:35 (c155f5830) MSP API: 1.46
# config rev: 53c4a3c
# board: manufacturer_id: SPRO, board_name: SPRACINGH7RF

# status
MCU H730 Clock=520MHz, Vref=3.31V, Core temp=56degC
Stack size: 2048, Stack address: 0x20020000
Configuration: CONFIGURED, size: 3927, max available: 65536
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked
GYRO=ICM42605, ACC=ICM42605, BARO=BMP388
OSD: MSP (30 x 13)
BUILD KEY: 5247170bc70d47594bc49eec6d8be5af (4.5.0)
System Uptime: 49 seconds, Current Time: 2024-04-29T10:23:15.024+00:00
CPU:43%, cycle time: 124, GYRO rate: 8064, RX rate: 500, System rate: 9
Voltage: 1196 * 0.01V (3S battery - OK)
I2C Errors: 9
SD card: Manufacturer 0x27, 30289920kB, 01/2005, v6.0, 'SD32G'
Filesystem: Ready
GPS: NOT ENABLED
Arming disable flags: CLI MSP NO_ACC_CAL
```

haslinghuis, 2024-04-29:
Cannot reproduce

```bash
# version
# Betaflight / STM32H730 (SP7R) 4.5.0 Apr 29 2024 / 10:43:20 (c155f5830) MSP API: 1.46
# config rev: 53c4a3c
# board: manufacturer_id: SPRO, board_name: SPRACINGH7RF

# status
MCU H730 Clock=520MHz, Vref=3.30V, Core temp=48degC
Stack size: 2048, Stack address: 0x20020000
Configuration: CONFIGURED, size: 3554, max available: 65536
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked
GYRO=ICM42688P, ACC=ICM42688P, BARO=BMP388
OSD: MSP (53 x 20)
BUILD KEY: 45e82ec2ee12bc127146fd2c723a8739 (4.5.0)
System Uptime: 126 seconds, Current Time: 2024-04-29T10:46:50.167+00:00
CPU:31%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 282 * 0.01V (0S battery - INIT)
I2C Errors: 9
SD card: Manufacturer 0x3, 31166976kB, 01/2007, v8.5, 'SD32G'
Filesystem: Ready
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP
```

[image]

hydra, 2024-04-29:
> Cannot reproduce

You're not using the same verson of the configurator.  `bae6c8fa` from your screenshot is an unknown commit here.

[image]

hydra, 2024-04-29:
Also, here's a build log, including removal of generated artifacts.

```
Hydra on DESKTOP-H1JUK44 in /cygdrive/d/Users/Hydra/Documents/dev/projects/betaflight/betaflight-configurator (master)
$ rm -rf cache debug debug node_modules
Hydra on DESKTOP-H1JUK44 in /cygdrive/d/Users/Hydra/Documents/dev/projects/betaflight/betaflight-configurator (master)
$ yarn install && yarn start
yarn install v1.22.21
[1/5] Validating package.json...
[2/5] Resolving packages...
warning Resolution field "libxmljs2@0.32.0" is incompatible with requested version "libxmljs2@^0.29.0"
[3/5] Fetching packages...
[4/5] Linking dependencies...
warning " > vite-plugin-pwa@0.17.5" has unmet peer dependency "workbox-build@^7.0.0".
warning " > vite-plugin-pwa@0.17.5" has unmet peer dependency "workbox-window@^7.0.0".
warning "@storybook/addon-actions > @storybook/addons@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > @storybook/addons@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > @storybook/api@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > @storybook/api@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > @storybook/components@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > @storybook/components@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > @storybook/theming@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > @storybook/theming@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-actions > react-inspector@5.1.1" has unmet peer dependency "react@^16.8.4 || ^17.0.0".
warning "@storybook/addon-links > @storybook/router@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-links > @storybook/router@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-essentials > @storybook/core-common@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-essentials > @storybook/core-common@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-essentials > @storybook/addon-docs > @mdx-js/react@1.6.22" has unmet peer dependency "react@^16.13.1 || ^17.0.0".
warning "@storybook/addon-essentials > @storybook/addon-docs > @storybook/preview-web@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-essentials > @storybook/addon-docs > @storybook/preview-web@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-essentials > @storybook/addon-docs > @storybook/source-loader@6.5.16" has unmet peer dependency "react@^16.8.0 || ^17.0.0 || ^18.0.0".
warning "@storybook/addon-essentials > @storybook/addon-docs > @storybook/source-loader@6.5.16" has unmet peer dependency "react-dom@^16.8.0 || ^17.0.0 || ^18.0.0".
warning " > babel-loader@8.3.0" has unmet peer dependency "webpack@>=2".
warning "@storybook/addon-essentials > @storybook/core-common > fork-ts-checker-webpack-plugin@6.5.3" has unmet peer dependency "typescript@>= 2.7".
warning " > @storybook/vue@6.5.16" has unmet peer dependency "css-loader@*".
warning "@storybook/vue > ts-loader@8.4.0" has unmet peer dependency "typescript@*".
warning "@storybook/vue > vue-docgen-loader > jscodeshift@0.13.1" has unmet peer dependency "@babel/preset-env@^7.1.6".
warning " > vue-loader@15.11.1" has unmet peer dependency "css-loader@*".
warning " > vue-loader@15.11.1" has unmet peer dependency "webpack@^3.0.0 || ^4.1.0 || ^5.0.0-0".
[5/5] Building fresh packages...
Done in 21.91s.
yarn run v1.22.21
$ run-script-os
$ set NODE_ENV=development && set NW_PRE_ARGS="--load-extension='./node_modules/nw-vue-devtools-prebuilt/extension' --remote-debugging-port=9222" && gulp debug
Building for platform(s): win64.
[15:36:55] Using gulpfile ~\Documents\dev\projects\betaflight\betaflight-configurator\gulpfile.js
[15:36:55] Starting 'debug'...
[15:36:55] Starting 'clean_dist'...
[15:36:56] Finished 'clean_dist' after 190 ms
[15:36:56] Starting 'process_package_debug'...
[15:36:56] Starting 'write_package_file'...
[15:36:56] Finished 'write_package_file' after 7.95 ms
[15:36:56] Finished 'process_package_debug' after 306 ms
[15:36:56] Starting 'dist_src'...
[15:36:56] Finished 'dist_src' after 393 ms
[15:36:56] Starting 'dist_node_modules_css'...
[15:36:58] Finished 'dist_node_modules_css' after 1.3 s
[15:36:58] Starting 'dist_ol_css'...
[15:36:58] Finished 'dist_ol_css' after 2.38 ms
[15:36:58] Starting 'dist_less'...
[15:36:58] Finished 'dist_less' after 503 ms
[15:36:58] Starting 'dist_locale'...
[15:36:58] Finished 'dist_locale' after 20 ms
[15:36:58] Starting 'dist_libraries'...
[15:36:58] Finished 'dist_libraries' after 5.14 ms
[15:36:58] Starting 'dist_resources'...
[15:36:58] Finished 'dist_resources' after 37 ms
[15:36:58] Starting 'dist_rollup'...
(node-resolve plugin) preferring built-in module 'util' over local alternative at 'D:\Users\Hydra\Documents\dev\projects\betaflight\betaflight-configurator\node_modules\util\util.js', pass 'preferBuiltins: false' to disable thi [... cut ...]
Create release folder in D:\Users\Hydra\Documents\dev\projects\betaflight\betaflight-configurator\debug\betaflight-configurator\win64
Update win64 executable icon
[15:37:41] Finished 'debug' after 31 s
[15:37:41] Starting 'post_build_none'...
[15:37:41] Finished 'post_build_none' after 166 μs
[15:37:41] Starting 'start_debug'...
Starting debug app (debug\betaflight-configurator\win64\betaflight-configurator.exe)...
[15:37:41] Finished 'start_debug' after 9.06 ms
[15:37:41] Finished 'debug' after 46 s
```

haslinghuis, 2024-04-29:
@hydra switched version - but still working

[image]

hydra, 2024-04-29:
It seems to be somehow related to the firmware, as I had an older FC which I hadn't updated to the latest firmware which only showed one telemetry slider.  I then updated it to `master`, after which I got the same double telemetry slider issue as in the OP.

I performed the above step using an installation of official build of BF and an official build of the BF configurator (from `betaflight-configurator_10.10.0_win64-installer.exe`) to rule out any build/development artifacts.

```
# version
# Betaflight / STM32H730 (SP7R) 4.5.0 Apr 29 2024 / 19:28:49 (c155f5830) MSP API: 1.46
# config rev: 2ceeadc
# board: manufacturer_id: SPRO, board_name: SPRACINGH7RF

# status
MCU H730 Clock=520MHz, Vref=3.27V, Core temp=51degC
Stack size: 2048, Stack address: 0x20020000
Configuration: CONFIGURED, size: 4297, max available: 65536
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked
GYRO=ICM42605, ACC=ICM42605, BARO=BMP388
OSD: MSP (30 x 13)
BUILD KEY: f16e3973633437b41470b27546c19bf3 (4.5.0)
System Uptime: 35 seconds, Current Time: 2024-04-29T19:30:54.172+00:00
CPU:33%, cycle time: 122, GYRO rate: 8196, RX rate: 15, System rate: 9
Voltage: 299 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 9
SD card: None inserted
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP
```

[image]

PR #3923 (2024-05-02): Fix dual telemetry in receiver tab for local builds
- Fixes #3920
closed 2024-05-02 by haslinghuis