betaflight/betaflight-configurator #3921: Mamba f435 2023A motors won't spin when arming

mlc76520, 2024-04-29:
### Describe the bug

Motors won't spin when arming. Same in BF motors tab; no spin. Works fine on 4.5.0 RC4.

### To Reproduce

Flash firmware. Spin motors in bf motors tab or arm kwad.

### Expected behavior

Motors should spin in bf motors tab. motors should spin when arming

### Configurator version

latest 10.10.0 c97deaf

### Flight controller configuration

```txt
# diff

# version
# Betaflight / AT32F435G (A435) 4.5.0 Apr 29 2024 / 13:59:50 (c155f5830) MSP API: 1.46
# config rev: 53c4a3c

# start the command batch
batch start

board_name MAMBAF435_2023A
manufacturer_id DIAT

# name: AKvert

# feature
feature TELEMETRY
feature OSD

# serial
serial 0 64 115200 57600 0 115200
serial 2 131073 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 3 1900 2100 0 0
aux 2 35 2 1900 2100 0 0
aux 3 36 1 1900 2100 0 0

# master
set acc_calibration = 2,-47,28,1
set max_check = 2000
set rc_smoothing_auto_factor = 140
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set vbat_min_cell_voltage = 350
set yaw_motors_reversed = ON
set small_angle = 180
set report_cell_voltage = ON
set osd_link_quality_pos = 3555
set osd_gps_speed_pos = 517
set osd_gps_lon_pos = 613
set osd_gps_lat_pos = 581
set osd_gps_sats_pos = 549
set osd_home_dir_pos = 183
set osd_home_dist_pos = 188
set osd_altitude_pos = 2533
set osd_warnings_pos = 14869
set osd_avg_cell_voltage_pos = 3587
set debug_mode = GYRO_SCALED
set craft_name = AKvert
set pilot_name = mlcFPV

profile 0

# profile 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 10

rateprofile 0

# rateprofile 0
set thr_mid = 30
set thr_expo = 50
set rates_type = BETAFLIGHT
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set throttle_limit_type = SCALE
set throttle_limit_percent = 90

# end the command batch
batch end

#
```

### Add any other context about the problem that you think might be relevant here

Works fine on 4.5.0 RC4

haslinghuis, 2024-04-29:
Do not open duplicate PR's for same issue as now we we have to look at two reports. Was going to move the issue to firmware.

closed 2024-04-29 by haslinghuis

mlc76520, 2024-04-29:
My apologies was not sure