betaflight/betaflight-configurator #3924: Issues connecting to BF via ELRS Wifi

txmonstr, 2024-04-30:
### Describe the bug

I have a problem connecting to my flightcontroller via the ELRS Wifi. It's a Radiomaster RP 3 with ELRS 3.3 and a Tmotor F7 FC with BF 4.4.1.

In the configurator 10.9 I have NO issues. Now I updated to configurator 10.10 and it doesn't wirk anymore. It connects to FC but is unable to ready the configuration. 

The FC is still at BF 4.4.1. Do I have to Update to 4.5 so I can connect via Wifi?

[image]
[image]
[image]

### To Reproduce

Connect to the Wifi of the ELRS reciever. In Betaflight configurator choose manual selection and use Port tcp://10.0.0.1. Klick on conntect. 

### Expected behavior

The configurator should conntect to the fc via the ELRS Wifi just like it does via a USB cable.

### Configurator version

10.10.0 (c97deaf)

### Flight controller configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.1 Apr 30 2023 / 16:19:25 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name TMOTORF7
manufacturer_id TMTR
mcu_id 001900145931500a20383539
signature 

# name: CL1 Build

# feature
feature -AIRMODE
feature ESC_SENSOR

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1950 2100 0 0
aux 1 1 4 1950 2100 0 0
aux 2 13 5 1950 2100 0 0
aux 3 28 4 900 1050 0 0
aux 4 35 2 1950 2100 0 0
aux 5 36 1 1950 2100 0 0
aux 6 39 6 900 1700 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 1.2

# vtx
vtx 0 6 0 0 1 1500 1700
vtx 1 6 0 0 2 1250 1500
vtx 2 6 0 0 3 1100 1250
vtx 3 6 0 0 4 950 1100

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 600
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_min_hz = 150
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 600
set acc_calibration = 57,39,-50,1
set min_check = 1005
set max_check = 1995
set rc_smoothing_auto_factor = 120
set dshot_idle_value = 400
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set bat_capacity = 1500
set vbat_scale = 111
set ibata_scale = 166
set battery_continue = ON
set yaw_motors_reversed = ON
set small_angle = 180
set gps_rescue_ascend_rate = 500
set gps_rescue_ground_speed = 500
set gps_rescue_descend_rate = 100
set gps_rescue_throttle_max = 1600
set gps_rescue_throttle_d = 15
set gps_rescue_velocity_i = 30
set gps_rescue_velocity_d = 20
set simplified_gyro_filter_multiplier = 120
set osd_warn_bitmask = 122879
set osd_rssi_alarm = 50
set osd_link_quality_alarm = 50
set osd_rssi_dbm_alarm = -95
set osd_cap_alarm = 1200
set osd_tim1 = 2561
set osd_tim2 = 2562
set osd_vbat_pos = 14721
set osd_rssi_pos = 57
set osd_link_quality_pos = 2104
set osd_rssi_dbm_pos = 2073
set osd_tim_1_pos = 14741
set osd_flymode_pos = 14348
set osd_throttle_pos = 14355
set osd_vtx_channel_pos = 30725
set osd_current_pos = 14728
set osd_mah_drawn_pos = 14735
set osd_pidrate_profile_pos = 14337
set osd_warnings_pos = 14569
set osd_disarmed_pos = 14634
set osd_flip_arrow_pos = 14445
set osd_rate_profile_name_pos = 33
set osd_pid_profile_name_pos = 37
set osd_rcchannels_pos = 129
set osd_stat_bitmask = 9586406
set osd_canvas_width = 30
set osd_canvas_height = 13
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5917
set vcd_video_system = NTSC
set pinio_box = 39,0,0,0
set stats_min_armed_time_s = 10
set stats_total_flights = 519
set stats_total_time_s = 78531
set stats_mah_used = 1143
set craft_name = CL1 Build

profile 0

# profile 0
set profile_name = Stock4.4
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set simplified_dterm_filter_multiplier = 120

profile 1

# profile 1
set profile_name = Testing
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set simplified_dterm_filter_multiplier = 120

profile 2

# profile 2
set profile_name = Akutuell
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set p_pitch = 61
set i_pitch = 120
set f_pitch = 137
set p_roll = 58
set i_roll = 114
set f_roll = 132
set p_yaw = 58
set i_yaw = 114
set f_yaw = 132
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set simplified_i_gain = 110
set simplified_pi_gain = 130
set simplified_feedforward_gain = 110
set simplified_dterm_filter_multiplier = 120

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rateprofile_name = Standart
set rates_type = BETAFLIGHT
set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 125
set roll_expo = 23
set pitch_expo = 23
set yaw_expo = 23
set roll_srate = 73
set pitch_srate = 75
set yaw_srate = 73

rateprofile 1

# rateprofile 1
set rateprofile_name = Cine thr
set roll_rc_rate = 5
set pitch_rc_rate = 5
set yaw_rc_rate = 5
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 40
set throttle_limit_type = SCALE
set throttle_limit_percent = 75

rateprofile 2

# rateprofile 2
set rateprofile_name = Cinefull
set roll_rc_rate = 5
set pitch_rc_rate = 5
set yaw_rc_rate = 5
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 40

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

ot0tot, 2024-04-30:
I can reproduce this with 10.10.0. 

Works fine with 10.9.0, does not work with 10.10.0 with the same error posted above. Also does not work with 10.10.0-RC2.

txmonstr, 2024-04-30:
Addition: it doesn't work with BF 4.5.0 and configurator 10.10.0 either. So it doesn't seem to be due to the version of the FC.

txmonstr, 2024-05-02:
I think I found the error. My FC version was 4.4.1. With this version it still doesn't work. If updated it to 4.5.0 and thought it still didn't work. But actually my mistake was I did forget to enable Serial Rx on UART 1. So logically the reciver could't talk to the FC. Once I enabled Serial Rx it works. The only "issue" is the gyro / accel icons on top aren't lit up. Just like they weren't aktivated. But apart from this I think everything works.

closed 2024-05-02 by txmonstr