betaflight/betaflight-configurator #3929: Font Manager button disabled

nerdCopter, 2024-05-02:
### Describe the bug

Font Manger button stuck in disabled state as reported by other pilots a few times. Experienced the same myself and typically use debug-build and just enable it.
Code either not setting or respecting `requires-max7456-font-device-detected`.
[image]
[image]
[image]

### To Reproduce

Flash target with `OSD (SD)` only. Maybe other combos, untested.

vcd_video_system = AUTO, NTSC, PAL does not matter.

### Expected behavior

Font Manager button enabled.

### Configurator version

10.10 release; 11.0 8ae2b050

### Flight controller configuration

```txt
# get vcd_video_system
vcd_video_system = AUTO
Allowed values: AUTO, PAL, NTSC, H
```

haslinghuis, 2024-05-02:
Can you check if it is set to `set osd_displayport_device = AUTO` somehow?

nerdCopter, 2024-05-03:
> Can you check if it is set to `set osd_displayport_device = AUTO` somehow?

```
defaults
```

```
# 

# get osd_displayport_device
osd_displayport_device = AUTO
Allowed values: NONE, AUTO, MAX7456, MSP, FRSKYOSD

# diff

# version
# Betaflight / STM32F405 (S405) 4.6.0 May  2 2024 / 12:55:20 (4640a5da9) MSP API: 1.46
# config rev: cdb9e49

# start the command batch
batch start

board_name FOXEERF405
manufacturer_id FOXE

# feature
feature TELEMETRY
feature LED_STRIP
feature OSD

# master
set osd_canvas_height = 13

profile 0

rateprofile 0

# end the command batch
batch end
```

[image]

<details> 
  <summary>dump</summary>
  

```
# dump

# version
# Betaflight / STM32F405 (S405) 4.6.0 May  2 2024 / 12:55:20 (4640a5da9) MSP API: 1.46
# config rev: cdb9e49

# start the command batch
batch start

board_name FOXEERF405
manufacturer_id FOXE

# name: -

# resources
resource BEEPER 1 A04
resource MOTOR 1 B00
resource MOTOR 2 B04
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 C06
resource MOTOR 6 C07
[... 1040 lines cut ...]
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
```
CONTENT
</details>

manual modified:
```
# get osd_displayport_device
osd_displayport_device = MAX7456
Allowed values: NONE, AUTO, MAX7456, MSP, FRSKYOSD
Default value: AUTO
```

[image]

haslinghuis, 2024-05-03:
Need to `set osd_displayport_device = MAX7456`

haslinghuis, 2024-05-03:
Did you try to power with battery?

```
"osdSetupNoOsdChipDetectWarning": {
        "message": "<span class=\"message-negative\"><b>WARNING:</b></span> No OSD chip was detected. Some flight controllers will not properly power the OSD chip unless connected to battery power. Please connect the battery before connecting the USB (PROPS REMOVED!)."
    },
```

nerdCopter, 2024-05-03:
that alert does not show; therefore i believe it detects the chip properly.
currently, i'm unsoldered, but in the past, i could upload fonts to this FC without lipo.  -- Although i cannot verify results, it processes upload-fonts from older UI. (progress bar)

nerdCopter, 2024-05-03:
i do now recall recently ApexF7 (couple months), i was forced to debug-mode-workraround as well, to perform the font/logo upload.

nerdCopter, 2024-05-04:
tested with LiPo, same result
[image]
[image]

haslinghuis, 2024-05-04:
Button is disabled here:

```bash
if (!OSD.data.state.isMax7456FontDeviceDetected || !OSD.data.state.haveMax7456FontDeviceConfigured) {
    $('.requires-max7456-font-device-detected').addClass('disabled');
}
```

where state would be
```bash
d.state = {};
d.state.haveSomeOsd = (d.flags !== 0);
d.state.haveMax7456Configured = bit_check(d.flags, 4);
d.state.haveFrSkyOSDConfigured = semver.gte(FC.CONFIG.apiVersion, API_VERSION_1_43) && bit_check(d.flags, 3);
d.state.haveMax7456FontDeviceConfigured = d.state.haveMax7456Configured || d.state.haveFrSkyOSDConfigured;
d.state.isMax7456FontDeviceDetected = bit_check(d.flags, 5) || (d.state.haveMax7456FontDeviceConfigured && semver.lt(FC.CONFIG.apiVersion, API_VERSION_1_43));
d.state.haveOsdFeature = bit_check(d.flags, 0);
d.state.isOsdSlave = bit_check(d.flags, 1);
d.state.isMspDevice = bit_check(d.flags, 6) && semver.gte(FC.CONFIG.apiVersion, API_VERSION_1_45);
```

nerdCopter, 2024-05-04:
[image]

haslinghuis, 2024-05-04:
Fresh flash on NEUTRONRCF435

[image]

[image]

haslinghuis, 2024-05-04:
@nerdCopter the issue seems to be `isMspDevice` is true in your case.

nerdCopter, 2024-05-06:
Considering i've seen two other reports of this, it is weird; however, i have a working button now.

I do not know if the source of the problem is potentially:
1) failing PG-reset, or
2) Configurator not 100% honoring full-chip-erase, or
3) a dying flash-chip.

I used my boot-button to force DFU mode rather than relying on reboot into DFU. I toggled the full-chip-erase a couple times to force it enabled.
Upon flashing, defaults worked as expected to enable the Font Manager button.
[image]
[image]

I also noticed similar weirdness when testing https://github.com/betaflight/betaflight-configurator/pull/3823 / https://github.com/betaflight/betaflight/pull/13404 where flashing did not present LED Strip option/tab even though it was enabled in the cloud-build. re-flashing the same hex resolved it.

closed 2024-05-06 by nerdCopter

nerdCopter, 2024-06-07:
still having trouble with this... 

### in the case of flashing both SD and HD,
what causes `isMax7456FontDeviceDetected` to be false when:
(`src/js/tabs/osd.js:2280:        d.state.isMax7456FontDeviceDetected = bit_check(d.flags, 5);`)
```
diff

# master
set osd_warnings_pos = 14740
set osd_displayport_device = MAX7456
set osd_canvas_width = 30
set osd_canvas_height = 13
set vcd_video_system = NTSC
set displayport_msp_fonts = 0,0,0,0
```
[image]

### in the case of flashing SD only, 
what causes `haveMax7456Configured` to be false? when:
```
# get osd_displayport_device
osd_displayport_device = AUTO
Allowed values: NONE, AUTO, MAX7456, MSP, FRSKYOSD

# get vcd_video_system
vcd_video_system = AUTO
Allowed values: AUTO, PAL, NTSC, HD
```
[image]

otherwise what causes `isMax7456FontDeviceDetected` to be false when:
```
diff

# master
set osd_displayport_device = MAX7456
set osd_canvas_height = 13
set vcd_video_system = NTSC
```
[image]

nerdCopter, 2024-06-07:
sorry, i have some clarity here
[image]

[PEBCAK](https://en.wiktionary.org/wiki/PEBCAK)

StadnikSindar, 2024-07-30:
> sorry, i have some clarity here ![image](https://private-user-images.githubusercontent.com/56646290/337756508-b0845ae0-713a-4ec3-84f5-1a85606672c8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIzNDI5ODMsIm5iZiI6MTcyMjM0MjY4MywicGF0aCI6Ii81NjY0NjI5MC8zMzc3NTY1MDgtYjA4NDVhZTAtNzEzYS00ZWMzLTg0ZjUtMWE4NTYwNjY3MmM4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzMwVDEyMzEyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1ZDU4MzExOTlmMmQyNmJkNmRhOGQzYTI4YWY3MzVjZjdlMzYxYTU4NmRkMGUwNTA1ZTY5OWQ3NmNmODY3MzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0I7l5YHnvJIOv0f0FqYfjIuiPfngW5KMM6Dhb4KgjZU)
> 
> [PEBKAC](https://en.wiktionary.org/wiki/PEBKAC)

It looks like that.  Now to be able to customize OSD I have to connect the battery  to drone before flashing new FW and keep an eye on VTX to not get overheated. Earlier it just went fine via USB.

Hhow it happened to work flawlessly in 4.4.3 version? Why this function is in fact degraded in newer version?

StadnikSindar, 2024-07-30:
Yes, exactly.  I work with SpeedyBee F405V3 and on 4.4.3 the OSD "font editor" was available after FW flashing via USB without battery. On 4.5.0 and 4.5.1 as well the  "font editor" button remains inactive uness you connect battery before flashing FW. And even in that case using battery does not guarantee. Sometimes it takes 2 attempts of flashing to make button  "font editor" become active.

quartit, 2024-08-29:
Got the same problem with 4.5.1, SpedeeBee F405 V3 and SpeedyBee TX800.
I tried to flash 4.5.0 and 4.5.1 with and without battery but no success. I confirm there is no such issue with 4.4.3.
However I managed to get it, not by flashing but by enabeling OSD on Configuration tab with battery + Save/Reboot.
Not sure about the exact combination, but after this, the OSD was working (not working before) and Font button unblocked.
Hope this could help other guys coming here.

RipperDrone, 2026-01-06:
Same issue on 2025.12.1 on 2 copters, device set to AUTO. I toggled OSD feature OFF<>ON several times, plugged LiPo first, USB second and vice versa. On one copter I finally got it working once, on the 2nd one (SucceX TwinG 40A AIO FC), it isn't working even after 15+ attempts :-(. Plus, I cannot see the OSD working, regardless which font I'm erxpecting..

RipperDrone, 2026-01-06:
@haslinghuis kind of solved it on the final 'stubbern' drone, as follows: I noticed that when closely following the Wiki, ie plugging in the LiPo first, connecting USB second, my smoke stopper triggered. Seems there's a ground loop / short existing on the FC. When doing it in the opposite order, I can connect both sources, though. However, with this sequence, the Font Manager button remains greyed out, regardless of re-flashing several times with full chip etasr, stepping down to v4. 5.3 or setting OSD type manually (changing from AUTO). Was hard to find, and funnily, the OSD never showed, as long as I kept the FC connected to USB. When powering the drone from a LiPo only, OSD is showing just normal 🤯. Thank you for your response, didn't expect it since this item has been closed long time ago. Just wanted to leave my case as a reference.. 💪❤️

haslinghuis, 2026-01-06:
@RipperDrone smoke stopper is good indication why it did not work for this board. Thanks for sharing experience.