betaflight/betaflight-configurator #3930: Holybro Kakute H7 (stack) on 4.5 gives me a "motor_protocol"error

x8n1620, 2024-05-02:
### Describe the bug

Same as  #3768, but with other FC
Motor protocol error in 4.5 using dshot600 on a  Holybro Kakute Stack (H7 FC and Holybro Tekko ESC)
I also try 
- dshot 300 or dshot150
- disable bidirectional dshot
- try it with bf-defaults, only aux and serial applied

Reflashed down to 4.4 and the quad works fine. Flashing 4.5 again, same problem

### To Reproduce

Flash a Kakute H7 Stack to BF 4.5

### Expected behavior

Arming Possible

### Configurator version

10.10.0

### Flight controller configuration

```txt
# 

# diff all

# version
# Betaflight / STM32H743 (SH74) 4.5.0 May  2 2024 / 14:43:13 (c155f5830) MSP API: 1.46
# config rev: cdb9e49

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEH7
manufacturer_id HBRO
mcu_id 004f003b3131511238363233
signature 

# name: DroehniV2

# feature
feature -AIRMODE
feature GPS
feature TELEMETRY
feature DISPLAY
feature OSD
feature ESC_SENSOR

# serial
serial 0 131073 115200 57600 0 115200
serial 3 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1825 2100 0 0
aux 1 1 2 1825 2100 0 0
aux 2 46 4 1825 2100 0 0
aux 3 13 5 1825 2100 0 0
aux 4 28 2 900 1175 0 0
aux 5 35 1 1825 2100 0 0

# rxfail
rxfail 0 s 1500
rxfail 1 s 1500
rxfail 2 s 1500
rxfail 3 s 1275
rxfail 4 s 2000
rxfail 7 s 1000

# master
set acc_calibration = 16,60,-348,1
set blackbox_sample_rate = 1/2
set dshot_bidir = ON
set bat_capacity = 1300
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set vbat_scale = 110
set ibata_scale = 245
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set report_cell_voltage = ON

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

x8n1620, 2024-05-03:
In addition I have to say, that I can spin the Motors from the Motor-Tab in BF. But arming is still not possibleFlashing again with Target: KAKUTEH/
Version 4.5.0 [28-Apr-2024]
Full Chip erease

Radio: CRSF
Motor: DSHOT
Other: Noting

*** Support data submitted *** 
Id: "b0031b20-37fd-48b7-b83b-959de233fa1e"

haslinghuis, 2024-05-03:
When not using Dshot telemetry try `set dshot_bitbang = OFF`

x8n1620, 2024-05-03:
After setting set dshot_bitbang = OFF
-> arming possible, Motors spin up

haslinghuis, 2024-05-03:
What firmware is the ESC running?

x8n1620, 2024-05-03:
Originaly yesterday 32.08, after having trouble to arm I update to the newest version 32.10

haslinghuis, 2024-05-03:
Using 32.8 here without problems. 32.10 did not work for me.
32.7 is still the recommended version:

https://betaflight.com/docs/wiki/getting-started/hardware/esc-firmware#blheli_32

x8n1620, 2024-05-03:
With the blheli suite I can't rollback to 32.7, only to 32.8.
After done this, the quad arms when I set dshot_bitbang = OFF
With dshot_bitbang = AUTO still no arming possible
What does this mean for my quad?

haslinghuis, 2024-05-03:
Take a look at `https://github.com/betaflight/betaflight/pull/7446`

x8n1620, 2024-05-03:
I did not understand a single word of what was said in #7446
But i will configure my quad and try (tomorrow).

Thanks for your patience and support!