betaflight/betaflight-configurator #3936: SOFTSERIAL not working after upgrade to 4.5 in BF Configurator 10.10.0 with FlywooF411

cpwilson, 2024-05-06:
### Describe the bug

Tried the 4.5 upgrade using Configurator 10.10.0 on Flywoo LR$ with FlywooF411 but unable to get SOFTSERIAL ports enabled so unable to control VTX which is using SOFTSERIAL1

### To Reproduce

Current firmware is 4.4.3 with FlywooF411 and enabling SOFTSERIAL works fine and able to use it for VTX.
Upgrade to 4.5 and no option to enable SOFTSERIAL in configurator.
Have tried multiple upgrades and changed resource mapping in CLI but still unable to get SOFTSERIAL to work.

This looks to be a similar issue to what I updated earlier under #13275 but that bug is closed.
Have tried adding it as an override option as suggested in #13305 but firmware will not compile and returns an error.

I get the feeling that SOFTSERIAL is possibly being dropped but it's unclear if that is the case or not?

### Expected behavior

If the FlywooF411 is still being supported in 4.5 then it's a fair assumption that it would be supported in full and not with some aspects, such as SOFTSERIAL, not supported.

SOFTSERIAL is needed as only have 2 UARTS

[image]

### Configurator version

10.10.0

### Flight controller configuration

```txt
Diff from 4.4.3

# diff all

# version
# Betaflight / STM32F411 (S411) 4.4.3 Apr 29 2024 / 22:29:44 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FLYWOOF411
manufacturer_id FLWO
mcu_id 002e00583030510431333433
signature 

# name: FLYWOO

# resources
resource MOTOR 2 B03
resource MOTOR 3 B10
resource MOTOR 4 A15
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 11 B06
resource SERIAL_RX 1 A10
resource SERIAL_RX 11 B07
resource ADC_BATT 1 B01
resource GYRO_EXTI 1 B05

# timer
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)

# dma
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6

# feature
feature GPS

# serial
serial 0 64 115200 57600 0 115200
serial 1 2 115200 115200 0 115200
serial 30 8192 115200 57600 0 115200

# mixer
mmix 0  1.000 -0.809  0.659 -1.000
mmix 1  1.000 -1.000 -0.659  1.000
mmix 2  1.000  0.809  0.659  1.000
mmix 3  1.000  1.000 -0.659 -1.000

# led
led 0 6,6::CO:8
led 1 7,6::CO:8
led 2 8,6::G:2
led 3 9,6::A:2

# mode_color
mode_color 6 0 2
mode_color 6 1 6

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 1900 2100 0 0
aux 2 2 1 1400 1600 0 0
aux 3 27 2 1900 2100 0 0
aux 4 46 3 1900 2100 0 0
aux 5 13 4 1400 1600 0 0
aux 6 19 5 1400 1600 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5999 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 50 100 200 MAX

# rxfail
rxfail 3 s 1300
rxfail 5 s 2000
rxfail 8 s 1500

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf1_dyn_min_hz = 0
set acc_trim_pitch = -6
set acc_trim_roll = 10
set acc_calibration = -95,-6,-52,1
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set align_board_pitch = 2
set bat_capacity = 1000
set small_angle = 30
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_rescue_alt_mode = FIXED_ALT
set gps_rescue_ground_speed = 1000
set gps_rescue_descent_dist = 40
set gps_rescue_descend_rate = 100
set gps_rescue_throttle_hover = 1300
set gps_rescue_allow_arming_without_fix = ON
set osd_units = IMPERIAL
set osd_cap_alarm = 700
set osd_alt_alarm = 1500
set osd_tim1 = 3
set osd_vbat_pos = 2496
set osd_rssi_pos = 2048
set osd_link_quality_pos = 2080
set osd_rssi_dbm_pos = 161
set osd_tim_1_pos = 2112
set osd_tim_2_pos = 321
set osd_flymode_pos = 2509
set osd_throttle_pos = 2368
set osd_vtx_channel_pos = 18439
set osd_crosshairs_pos = 2317
set osd_ah_pos = 2190
set osd_current_pos = 2070
set osd_mah_drawn_pos = 2104
set osd_craft_name_pos = 33
set osd_gps_speed_pos = 2375
set osd_gps_lon_pos = 76
set osd_gps_lat_pos = 44
set osd_gps_sats_pos = 2553
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2093
set osd_flight_dist_pos = 184
set osd_compass_bar_pos = 2539
set osd_altitude_pos = 18770
set osd_warnings_pos = 2442
set osd_avg_cell_voltage_pos = 2528
set osd_disarmed_pos = 363
set osd_flip_arrow_pos = 65
set osd_core_temp_pos = 248
set osd_log_status_pos = 97
set osd_stat_bitmask = 16154
set osd_canvas_width = 30
set osd_canvas_height = 16
set vtx_band = 4
set vtx_channel = 7
set vtx_power = 5
set vtx_low_power_disarm = ON
set vtx_freq = 5860
set vcd_video_system = PAL
set gyro_1_align_yaw = 1800
set craft_name = FLYWOO

profile 0

# profile 0
set vbat_sag_compensation = 100

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

cpwilson, 2024-05-06:
Apologies, if this is a similar/duplicate of #13275 but would be good to understand if this is a actually a BUG, something I am doing wrong, or if it is that SOFTSERIAL is no longer going to be supported.

haslinghuis, 2024-05-06:
Duplicate of https://github.com/betaflight/betaflight/issues/13275

Please keep in #13275 as this is not a configurator issue.

closed 2024-05-06 by haslinghuis

cpwilson, 2024-05-06:
#13275 is Closed?

So will it we get a fix or is softserial no longer being supported?

haslinghuis, 2024-05-06:
Might join our [discord](https://discord.com/invite/gV4XSq3fus)