betaflight/betaflight-configurator #3943: BlackBox Tab: Remove "Save File to file button"

Canaill51, 2024-05-09:
### Is your feature request related to a problem? Please describe

Hi team,
Regarding to this button "Save File to file" in Blackbox tab, is there a specific reason that it still does exist?

Actually, reading at its comment, seems like it is not recommended to use it. The way to go is "Activate Mass Storage Device Mode" button.

Maybe removing this button would be a good idea?

Thanks to this, users would not have to choice which button to use, as there is one button only available.

### Describe the solution you'd like

Remove "Save File to file" button from Blackbox Tab.

### Describe alternatives you've considered

Use the existing and recommended "Activate Mass Storage Device Mode" button.

Canaill51, 2024-05-09:
[image]

haslinghuis, 2024-06-03:
I am still using that feature during development on linux as I'm lazy to use the Mass Storage Mode.

haslinghuis, 2024-06-03:
Maybe could make it expert only option first. Or perhaps experimental option.

PR #4056 (2024-06-23): Make msc-not-support expert only
- resolves #3943

Expert mode disabled
[image]

Expert mode enabled
[image]
closed 2024-06-23 by haslinghuis