betaflight/betaflight-configurator #3969: remove gconf2 dependency from linux builds

pitts-mo, 2024-05-17:
### Is your feature request related to a problem? Please describe

betaflight-configurator 10.10.0 can not be installed and does not work properly on current linux systems due to dependance on depretiated gconf2.

### Describe the solution you'd like

remove gconf2 dependency from linux builds

### Describe alternatives you've considered

There has been partial success with suggestions mentioned under issues #2950  and #3860.

pitts-mo, 2024-05-17:
Thank you @haslinghuis ,
I had noticed the readme update along with the notes on the two other issues. However, I am still getting hung up on attempting to install betaflight-configurator on ubuntu 24.04. 

I am leary of adding the old 23.04 lunar universe repo as I have had issues attempting to use old repos as a workaround in the past.

-p

My system:
$ lsb_release -a 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble

$ uname -a 
Linux h170m 6.8.0-31-generic #31-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 00:40:06 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

Readme step resuts:
$ sudo usermod -aG dialout ${USER}

$ sudo mkdir /usr/share/desktop-directories/
mkdir: cannot create directory ‘/usr/share/desktop-directories/’: File exists

$ sudo apt install libatomic1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libatomic1 is already the newest version (14-20240412-0ubuntu1).
libatomic1 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

$ sudo dpkg -i betaflight-configurator_10.10.0_amd64.deb 
Selecting previously unselected package betaflight-configurator.
(Reading database ... 246121 files and directories currently installed.)
Preparing to unpack betaflight-configurator_10.10.0_amd64.deb ...
Unpacking betaflight-configurator (10.10.0) ...
dpkg: dependency problems prevent configuration of betaflight-configurator:
 betaflight-configurator depends on libgconf-2-4; however:
  Package libgconf-2-4 is not installed.

dpkg: error processing package betaflight-configurator (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 betaflight-configurator

hegelocampus, 2024-06-15:
Given gconf has been deprecated for 13 years, what the blockers are to moving to dconf?

nerdCopter, 2024-06-15:
> Given gconf has been deprecated for 13 years, what the blockers are to moving to dconf? Having users install an old lib feels like a fragile bandaid here.

NWjs (Chromium) based installation are no longer applicable. PWA (Progressive Web App) is in development moving forward.  https://app.betaflight.com

closed 2024-06-27 by haslinghuis