betaflight/betaflight-configurator #3989: "Reorder motors" wizard uses to small idleThrottleValue

shukari, 2024-05-30:
### Describe the bug

No motor is spinning in the "Reorder motors" wizard because it uses 1060 as idleThrottleValue but my motors start to spin at 1070.
[Code line](https://github.com/limonspb/betaflight-configurator/blob/43d95faf5115a24bf5c7d2cbb995f0b66030ffa5/src/js/tabs/motors.js#L1231)

### To Reproduce

1. open "Reorder motors" wizard
2. accept warning
3. you should select the spinning motor
4. no motor is spinning

### Expected behavior

a motor should spinning

### Configurator version

10.10.0

### Add any other context about the problem that you think might be relevant here

It would be nice if there would be a cli setting to change that value or it should use the idle setting

haslinghuis, 2024-06-03:
@atomgomba @limonspb should we revive #2766

atomgomba, 2024-06-03:
> @atomgomba @limonspb should we revive #2766

Though not sure how much the base branch has diverged in the meantime, but I would gladly help implementing a UI widget for this again

haslinghuis, 2024-06-03:
A ... lot :rofl:  - but reorder motors is somewhat isolated from refactoring.

limonspb, 2024-06-04:
that be neat to have... Maybe just in the general settings? Since it's very unusual thing to twek

YarosMallorca, 2024-08-31:
Why not use the static idle value for this?

haslinghuis, 2024-08-31:
Or dynamic value

jazzroutine, 2024-09-03:
And for me interface after hitting "reorder motors" or "motor directions" is just blank window. Only in browser. 
In dedicated Windows app or on Android phone everything is fine. 
[image]

closed 2024-09-23 by haslinghuis