betaflight/betaflight-configurator #3994: fail safe is accidentally activated?!

upyopyo, 2024-05-31:
### Describe the bug

Hi, I was testing a mode switch in my home, second floor.
BF config 10.10.0
Fireware 4.5.0 GepRC F222 AIO

Basically I was doing:
1 turn on arming
2 turn on fail safe mode (AUX4)
3 drone is deactivated/disarmed.
This is fine result as I expect. 

However,
1 turn on arming
2 rise up throttle a bit, but drone was still on ground
3 turn on fail safe mode swtich (AUX4), then drone is suddenly jump up and hit a ceilling. I think fail safe is activated. 
This result does not make a sense because **the minimum distance to home** is set to 10m, so all motors should be stopped, instead of activating a fail safe mode. I believe this never happened before updating a new fireware.

[image]
[image]

### To Reproduce

1 turn on arming
2 rise up throttle a bit, but drone was still on ground
3 turn on fail safe mode swtich (AUX4), then drone is suddenly jump up because I believe the fail safe mode is activated. 
(4 I even took off flip over after crash mode from AUX4, but it happened again)

### Expected behavior

1 turn on arming
2 rise up throttle a bit, but drone was still on ground
3 turn on fail safe mode swtich (AUX4), then all motors are stopped because the drone is within 10m. At least, I believe my drone with old BF version stopped all motors, even though it's flying in air 5m away, and dropped on ground as correct behaiver. 

### Configurator version

10.10.0

### Flight controller configuration

```txt
2024-05-31 @22:05:48 -- OS: Windows

2024-05-31 @22:05:48 -- Configurator: 10.10.0 (c97deaf)

2024-05-31 @22:05:50 -- Chrome API Error: Failed to connect to the port..

2024-05-31 @22:05:50 -- Failed to open serial port

2024-05-31 @22:05:52 -- Serial port successfully opened with ID: 1

2024-05-31 @22:05:52 -- MultiWii API version: 1.46.0

2024-05-31 @22:05:52 -- Flight controller info, identifier: BTFL, version: 4.5.0

2024-05-31 @22:05:52 -- Running firmware released on: Oct 6 2023 23:04:09

2024-05-31 @22:05:52 -- Board: GEPR/GEPRC_F722_AIO(STM32F7X2), version: 0

2024-05-31 @22:05:52 -- Unique device ID: 0x2800305333500820333931

2024-05-31 @22:05:52 -- Build Key: af30cc7cea034a138a3f69cff64252d3

2024-05-31 @22:05:52 -- Craft name: mmu 4inch F722

2024-05-31 @22:05:52 -- Arming Disabled

2024-05-31 @22:06:49 -- EEPROM saved

2024-05-31 @22:07:04 -- Serial port successfully closed

2024-05-31 @22:14:31 -- Serial port successfully opened with ID: 2

2024-05-31 @22:14:31 -- MultiWii API version: 1.46.0

2024-05-31 @22:14:31 -- Flight controller info, identifier: BTFL, version: 4.5.0

2024-05-31 @22:14:31 -- Running firmware released on: Oct 6 2023 23:04:09

2024-05-31 @22:14:31 -- Board: GEPR/GEPRC_F722_AIO(STM32F7X2), version: 0

2024-05-31 @22:14:31 -- Unique device ID: 0x2800305333500820333931

2024-05-31 @22:14:31 -- Build Key: af30cc7cea034a138a3f69cff64252d3

2024-05-31 @22:14:31 -- Craft name: mmu 4inch F722

2024-05-31 @22:14:31 -- Arming Disabled

2024-05-31 @22:14:40 -- CLI mode detected

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Oct  6 2023 / 23:04:09 (63f506530) MSP API: 1.46
# config rev: 5d34e80

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# name: mmu 4inch F722

# feature
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 2 115200 115200 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# led
led 0 2,9::CO:3
led 1 3,9::CO:3
led 2 4,9::CT:1
led 3 5,9::CT:1
led 4 6,9::CT:1
led 5 7,9::CT:1
led 6 8,9::CT:1
led 7 9,9::CT:1
led 8 10,9::CO:3
led 9 11,9::CO:3

# color
color 1 0,255,156
color 2 203,0,153
color 3 224,0,94
color 8 83,0,255
color 10 0,0,49
color 11 270,255,255
color 12 95,0,255
color 14 0,255,24
color 15 242,0,255

# mode_color
mode_color 7 0 7

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 2 2 900 1300 0 0
aux 3 46 3 1700 2100 0 0
aux 4 13 4 900 1375 0 0
aux 5 15 0 900 1300 0 0
aux 6 28 2 1300 1700 0 0
aux 7 35 3 2000 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lpf1_static_hz = 300
set gyro_lpf2_static_hz = 600
set dyn_notch_q = 120
set dyn_notch_min_hz = 150
set gyro_lpf1_dyn_min_hz = 300
set gyro_lpf1_dyn_max_hz = 600
set acc_lpf_hz = 10
set acc_calibration = 8,-44,-30,1
set baro_hardware = NONE
set rssi_channel = 16
set rssi_scale = 200
set rssi_offset = -100
set serialrx_provider = CRSF
set dshot_idle_value = 200
set failsafe_delay = 4
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set failsafe_recovery_delay = 20
set align_board_yaw = 90
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = MSAS
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_min_start_dist = 10
set gps_rescue_alt_mode = FIXED_ALT
set gps_rescue_ascend_rate = 500
set gps_rescue_ground_speed = 2000
set gps_rescue_descent_dist = 200
set gps_rescue_landing_alt = 5
set gps_rescue_throttle_max = 1600
set gps_rescue_throttle_hover = 1250
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_throttle_p = 150
set gps_rescue_throttle_i = 20
set gps_rescue_throttle_d = 50
set gps_rescue_velocity_p = 80
set gps_rescue_velocity_i = 20
set gps_rescue_velocity_d = 15
set gps_rescue_yaw_p = 40
set simplified_gyro_filter_multiplier = 120
set osd_vbat_pos = 3529
set osd_rssi_pos = 3497
set osd_link_quality_pos = 3369
set osd_rssi_dbm_pos = 3401
set osd_flymode_pos = 3465
set osd_throttle_pos = 2248
set osd_mah_drawn_pos = 165
set osd_craft_name_pos = 2679
set osd_gps_speed_pos = 2216
set osd_gps_lon_pos = 2150
set osd_gps_lat_pos = 2118
set osd_gps_sats_pos = 3209
set osd_home_dir_pos = 7271
set osd_home_dist_pos = 7273
set osd_altitude_pos = 2282
set osd_warnings_pos = 14808
set osd_avg_cell_voltage_pos = 3561
set osd_pit_ang_pos = 133
set osd_rol_ang_pos = 101
set osd_disarmed_pos = 2393
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set gyro_2_bustype = SPI
set gyro_2_sensor_align = CW0
set craft_name = mmu 4inch F722

profile 2

# profile 2
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set iterm_windup = 100
set yaw_lowpass_hz = 0
set p_pitch = 46
set i_pitch = 90
set d_pitch = 38
set f_pitch = 95
set p_roll = 42
set i_roll = 85
set d_roll = 35
set f_roll = 90
set i_yaw = 90
set f_yaw = 90
set horizon_level_strength = 50
set d_min_roll = 23
set d_min_pitch = 25
set simplified_pids_mode = OFF
set simplified_dterm_filter_multiplier = 120

rateprofile 0

# rateprofile 0
set thr_mid = 20
set thr_expo = 80
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 50
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end
```

### Add any other context about the problem that you think might be relevant here

my FC is firmware 4.5.0 GepRC F222 AIO.
If my idea is wrong, could you please tell me a link that explains how to set up GPS rescue in BF 10.10.0 in detail? This really freaked me out. Thanks.

haslinghuis, 2024-06-03:
GPS Rescue has changed a lot. Please read updated [documentation](https://betaflight.com/docs/wiki/guides/current/GPS-Rescue-v4-5)

[image]

closed 2024-06-03 by haslinghuis

upyopyo, 2024-06-04:
yes, I removed 'FLIP OVER AFTER CRASH' mode from failsafe switch, but same problem happened, I hope someone reports the same issue, then developpers could work on the issue.