betaflight/betaflight-configurator #4032: Does not work with version 4.6

Samodelkin-YouTube, 2024-06-17:
### Describe the bug

Compiled firmware 4.6 for my own DIY FC. After that, the configurator does not want to start
[image]

### To Reproduce

\\\\\

### Expected behavior

Launching the configurator and setting up the flight controller

### Configurator version

10.10.0

haslinghuis, 2024-06-17:
*For 4.6 we need https://app.betaflight.com. We had much support issues where people were using an older configurator with newer firmware. This way we also gain early feedback from users using firmware 4.6*

Samodelkin-YouTube, 2024-06-17:
> _For 4.6 we need https://app.betaflight.com. We had much support issues where people were using an older configurator with newer firmware. This way we also gain early feedback from users using firmware 4.6_
Does it only work with 4.6?  Is it possible to work with old versions?
Also I couldn't find version 11 of the configurator.

haslinghuis, 2024-06-17:
11.0 supports firmware 4.3 - 4.6

Samodelkin-YouTube, 2024-06-17:
OK. Thanks

YarosMallorca, 2024-06-21:
@haslinghuis Where is the github repo for the 11.0 Betaflight Configurator?
Can't find the source code for it.

haslinghuis, 2024-06-21:
@YarosMallorca you already have it forked :upside_down_face:

haslinghuis, 2024-06-23:
Try chrome or chromium.

For Ubuntu it needs an non sand boxed installation - not the snap version.

YarosMallorca, 2024-07-07:
Currently not supported on browsers that aren't Chrome, Edge and Opera.

Check this compatibility list: https://developer.mozilla.org/en-US/docs/Web/API/Web_Serial_API#browser_compatibility

closed 2024-08-15 by github-actions[bot]

Yamboss, 2024-11-26:
still can't find a betaflight configurator for 4.6 and the web don't work on my mac... i really want to test the alt mode on the new 4.6 i was waiting for it.

YarosMallorca, 2024-11-26:
@Yamboss Make sure you use Chrome, the Betaflight PWA version (online) works, can confirm on my MacBook Air M3.