betaflight/betaflight-configurator #4065: Ubuntu-debian version:  betaflight-configurator_10.10.0_amd64.deb

steel-allendale, 2024-06-26:
### Is your feature request related to a problem? Please describe

I am using an arm processor and there is a compatibility problem

### Describe the solution you'd like

A download compatible with arm64

### Describe alternatives you've considered

I don't see any workarounds

### Other information

dpkg: error processing archive betaflight-configurator_10.10.0_amd64.deb (--install):
 package architecture (amd64) does not match system (arm64)
Errors were encountered while processing:
 betaflight-configurator_10.10.0_amd64.deb

haslinghuis, 2024-06-26:
We moved on to the cloud for configuration and blackbox log viewer. Please try the new [PWA](https://app.betaflight.com).

closed 2024-06-26 by haslinghuis

steel-allendale, 2024-06-26:
Thanks Mark.  That’s great.
Best wishes
Michael