betaflight/betaflight-configurator #4084: Preset Save Backup not working

simoccc, 2024-07-04:
### Describe the bug

Preset Save Backup not working 

### To Reproduce

go to presets , click save backup , check the file saved and I got :

defaults nosave

$M&ltlldiff all
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###

### Expected behavior

normally saved backup of FC configuration 

### Configurator version

10.10.0

### Flight controller configuration

```txt
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Jul  3 2024 / 20:29:22 (c155f5830) MSP API: 1.46
# config rev: e167cfd

# start the command batch
batch start

board_name BLADE_F7
manufacturer_id RUSH

# feature
feature TELEMETRY
feature OSD

# serial
serial 0 64 115200 57600 0 115200
serial 1 131073 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 35 2 1700 2100 0 0
aux 3 40 4 1700 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 41,14,29,1
set baro_hardware = NONE
set dshot_bidir = ON
set motor_output_reordering = 1,0,3,2,4,5,6,7
set osd_vbat_pos = 2124
set osd_rssi_pos = 2156
set osd_link_quality_pos = 2219
set osd_tim_1_pos = 2484
set osd_flymode_pos = 2132
set osd_current_pos = 2401
set osd_mah_drawn_pos = 2432
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2114
set debug_mode = FFT_FREQ

profile 0

# profile 0
set d_pitch = 34
set d_roll = 30
set simplified_dmax_gain = 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 112
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 73

# end the command batch
batch end
```

### Add any other context about the problem that you think might be relevant here

got this error in different FC and builds , cannot find a specific pattern. happened in MacOS latest Apple silicon processor OS. I've seen it even in one of my friend's Windows 10 PC too.

Canaill51, 2024-07-05:
Strange this issue, I did build, flashed, configured and backed-up this exactly same FC on my computers (laptop windows 10 & desktop Windows 11), and worked fine.

However, I experienced this kind of strange behavior in the past, whereas file was pretty empty...
But not now, works pretty well.

simoccc, 2024-07-05:
I thought it was a MacOS issue but when I saw it on Windows too...

Tested now with 3 different builds one after the other and it worked, even the Rush one I posted on this issue. Same connection and cables as when there was the issue. Seems random to me. I haven't checked the event log of configurator though, I forgot. 

Maybe a check and visual warning if backup is consistent can be added to configurator ? 

As I said before I saw it happened in a friend laptop windows 10 too , on the field . We were trying to update FC firmware on the fly , preset backup save , updated the FC normally, and after that we wanted to import some settings from the backup but the file was empty.

Canaill51, 2024-07-05:
> 
> As I said before I saw it happened in a friend laptop windows 10 too , on the field . We were trying to update FC firmware on the fly , preset backup save , updated the FC normally, and after that we wanted to import some settings from the backup but the file was empty.
> 
> 

I did experience exactly same behavior in the past... 
Not anymore... I don't know what happened neither.

haslinghuis, 2024-07-05:
Next time it happens try

1. replugging flight controller
2. restarting configurator
3. rebooting computer

closed 2024-08-12 by github-actions[bot]