betaflight/betaflight-configurator #4110: Easy language change after installation

szafranski, 2024-07-28:
### Is your feature request related to a problem? Please describe

I cannot change the language of the BF configurator. 

### Describe the solution you'd like

Drop-down menu in the settings tab

### Describe alternatives you've considered

Change during reinstallation

### Other information

Sorry if that's something that seems stupid but I tried to find the solution and even re-installation with clearing the installation path is not seeming to work. Thanks

haslinghuis, 2024-07-28:
Language can be changed at the bottom of the Welcome tab.

j4k0xb, 2024-09-18:
The Welcome tab is mainly for displaying information, and I didn't expect a setting there (having to scroll down to see it made it even harder to find).
imo it makes more sense to put it in the options tab

PR #4174 (2024-09-30): Add language option
- fixes #4110 
- adds language selection to options tab
- keeps language selection on landing tab.
- no refresh required
- For future improvement an icon could improve UX experience
closed 2024-09-30 by haslinghuis