betaflight/betaflight-configurator #4116: BF4.5 Arming disable flags  "NO_ACC_CAL" display wrong value "DSHOT_BITBANG" in betaflight-configurator

kedeng, 2024-07-30:
### Describe the bug

BF4.5 disable flags "NO_ACC_CAL"  render wrong value "DSHOT_BITBANG" in betaflight-configurator System Info panel.

[image]

### To Reproduce

1. Update FC firmware to 4.5.0 
2. Goto Modes  --> Add Range for ANGLE mode
3. Goto Setup, Check "Arming Disable Flags" in "System info" panel
        -----> you can see "RX_FAILSAFE MSP DSHOT_BITBANG"
5. Goto CLI and send "status", check cli result
        -----> you can see "Arming disable flags: RXLOSS CLI MSP NO_ACC_CAL"``

### Expected behavior

Disable Flags "DSHOT_BITBANG" expected to "NO_ACC_CAL"

### Configurator version

10.10.0

haslinghuis, 2024-07-30:
Fixed in #4030

kedeng, 2024-07-30:
> Fixed in #4030

Not Fixed, can still be reproduced in latest build.

In BF https://github.com/betaflight/betaflight/blob/6e0855c296594f644d714227e940133685d732f3/src/main/fc/runtime_config.c#L36
```
const char *armingDisableFlagNames[]= {
    "NOGYRO",
    "FAILSAFE",
    "RXLOSS",
    "NOT_DISARMED",
    "BOXFAILSAFE",
    "RUNAWAY",
    "CRASH",
    "THROTTLE",
    "ANGLE",
    "BOOTGRACE",
    "NOPREARM",
    "LOAD",
    "CALIB",
    "CLI",
    "CMS",
    "BST",
    "MSP",
    "PARALYZE",
    "GPS",
    "RESCUE_SW",
    "DSHOT_TELEM",
    "REBOOT_REQD",
    "DSHOT_BBANG",
    "NO_ACC_CAL",
    "MOTOR_PROTO",
    "ARMSWITCH",
};
```

In betaflight-configurator https://github.com/betaflight/betaflight-configurator/blob/d328813d80a56936ba51e544e7f943cb2edbd7e8/src/js/tabs/setup.js

```
            let disarmFlagElements = [
                'NO_GYRO',
                'FAILSAFE',
                'RX_FAILSAFE',
                'NOT_DISARMED',
                'BOXFAILSAFE',
                'RUNAWAY_TAKEOFF',
                'CRASH_DETECTED',
                'THROTTLE',
                'ANGLE',
                'BOOT_GRACE_TIME',
                'NOPREARM',
                'LOAD',
                'CALIBRATING',
                'CLI',
                'CMS_MENU',
                'BST',
                'MSP',
                'PARALYZE',
                'GPS',
                'RESC',
                'RPMFILTER',
                // Introduced in 1.42
                'REBOOT_REQUIRED',
                'DSHOT_BITBANG',
                // Introduced in 1.43
                'ACC_CALIBRATION',
                'MOTOR_PROTOCOL',
                // 'ARM_SWITCH',           // Needs to be the last element, since it's always activated if one of the others is active when arming
            ];

            if (semver.gte(FC.CONFIG.apiVersion, API_VERSION_1_46)) {
                disarmFlagElements.splice(disarmFlagElements.indexOf('RPMFILTER'), 0, 'DSHOT_TELEM');
            }
```

If API_VERSION_1_46 is true, disarmFlagElements  is: 

```
[
	"NO_GYRO",
	"FAILSAFE",
	"RX_FAILSAFE",
	"NOT_DISARMED",
	"BOXFAILSAFE",
	"RUNAWAY_TAKEOFF",
	"CRASH_DETECTED",
	"THROTTLE",
	"ANGLE",
	"BOOT_GRACE_TIME",
	"NOPREARM",
	"LOAD",
	"CALIBRATING",
	"CLI",
	"CMS_MENU",
	"BST",
	"MSP",
	"PARALYZE",
	"GPS",
	"RESC",
	"DSHOT_TELEM",
	"RPMFILTER",
	"REBOOT_REQUIRED",
	"DSHOT_BITBANG",
	"ACC_CALIBRATION",
	"MOTOR_PROTOCOL"
]
```

"RPMFILTER" is not in armingDisableFlagNames.

[image]

PR #4118 (2024-08-03): Fix usage of splice
- fixes #4116
closed 2024-08-03 by VitroidFPV