betaflight/betaflight-configurator #4126: Minimum Voltage Should Mean Minimum Voltage

jefffhaynes, 2024-08-11:
### Describe the bug

Perplexingly, the Minimum Voltage setting seems to mean "critical warning voltage" setting.  Being new to this, I didn't understand what the great variety of different alarms meant and ruined a battery, thinking that this setting would protect me from dropping below the minimum voltage.

Ideally, this would function as described.  However, if that's not deemed appropriate or possible, then the label on this setting should be changed to reflect the actual function of the setting.

### To Reproduce

Open the Configurator.  Click on Power/Battery tab.

### Expected behavior

To prevent battery from dropping below minimum voltage by going into failsafe.

### Configurator version

10.10

nerdCopter, 2024-08-12:
That warning should have displayed a "LAND NOW" in the OSD.  I believe it also prevents arming if under such voltage, but such needs to be confirmed.

How low did you go, LiPo should not be "ruined" but maybe has a little less life in it.  Unless it's a 1S, then yes, maybe ruined.

Please follow this chart https://blog.ampow.com/lipo-voltage-chart/, but the defaults are conservative and safe.

jefffhaynes, 2024-08-12:
I'm not using an OSD, I'm developing a companion controller. I'm at 6.2V on a 3S. 

I'm not clear how this really changes the fact that the label is essentially incorrect though.

nerdCopter, 2024-08-12:
yeah, maybe it can be reworded. as it sits now the firmware source code uses both `vbatmincellvoltage` (per cell) and `batteryCriticalVoltage` (whole pack) in relation to this.  the configurator relates to the first variable, and thus was intended to match.  maybe a new help-tip message is an alternative.

off subject, may be one could use RX/TX telemetry to report voltage.  No way for the flight-controller to stop power consumption automatically however.

jefffhaynes, 2024-08-12:
I'm no expert (clearly), but why couldn't failsafe be an option on low voltage?

nerdCopter, 2024-08-12:
> I'm no expert (clearly), but why couldn't failsafe be an option on low voltage? I realize that would be a FC change.

pilots definitely prefer to risk a LiPo over losing a quad, this is why firmware will not disarm based on low voltage. 

I'll invite more senior devs to comment at their leisure as well. potentially it could be added as some failsafe option, but i dont know that the idea has ever come up before.  I see more and more people trying to automate betaflight, but this was not its intention. its purpose is for manual piloting.  Ardupilot and possibly others are designed with automation in mind. but this is just my opinion.

jefffhaynes, 2024-08-12:
Well if there's one takeaway, I would say it's consider changing the label to "LAND NOW alert" or something. Rather than a cryptic description, just tell it like it is and remove the ambiguity.

nerdCopter, 2024-08-12:
PR created. open to comments. https://github.com/betaflight/betaflight-configurator/pull/4129

closed 2024-08-16 by blckmn