betaflight/betaflight-configurator #4145: Ubuntu 24.04 LTS missing dependency

hremling, 2024-08-29:
### Describe the bug

The installation of betaflight configurator on Ubuntu 24.04  LTS fails due to a missing dependency. A Paket called libgconf-2-4 has no install candidates in the official noble nombat repositories. I researched the issue and found some forum posts, saying that libgconf-2-4  deprecated. 

A workaround might be to add the 22.04 LTS repositories to the sources.list file and install the missing packet manually. 

However it should be reviewed, if the dependency is really necessary or another package could be used instead. 

### To Reproduce

Ubuntu 24.04.1 LTS Clean Install

Download betaflight-configurator_10.10.0_amd64.deb file

dpkg -I betaflight-configurator_10.10.0_amd64.deb

### Expected behavior

All dependencies are installed automatically from the default Ubuntu 24.04 repository without errors 

### Configurator version

10.10.0

nerdCopter, 2024-08-30:
duplicate
https://github.com/betaflight/betaflight-configurator/issues/3969
https://github.com/betaflight/betaflight-configurator/issues/3869
https://github.com/betaflight/betaflight-configurator/issues/2950

shahmirthesquid, 2024-09-24:
Hello, you can fix this issue by manually installing the following packages:

gconf2-common_3.2.6-5_all.deb
libgconf-2-4_3.2.6-5_amd64.deb 

Google these package names, download them, and install. I downloaded them from debian pkgs org. Then you will be able to install betaflight with no problem. This was working for me on Linux Mint 22 and 23.

After installing betaflight I did the USB permissions command like it says in the betaflight wiki for linux, and everything was connecting perfectly.

nerdCopter, 2024-09-24:
i feel like using the portable (.zip) is a functional "fix" as well.  however, the included `.desktop` file should be edited to point to the correct location of the executable (not `/opt/...`).

hremling, 2024-09-24:
I think both of those options shouldn't be called a "fix" because they require manual intervention from the user. In addition, those workarounds could stop working at any time because they rely on the deprecated libgconf-2-4 package. Even if that's not happening anytime soon, it poses a security risk at least.
My issue was just intended to give a heads up to the developers since the Ubuntu 24.04.1 LTS was being rolled out to the users. As one of the most widely used Linux distributions, this might have lead to many more users having to face the same issue.

nerdCopter, 2024-09-24:
understood. Binary installables are no longer in the Betaflight workflow in favor of https://app.betaflight.com

shahmirthesquid, 2024-09-26:
Could we please add some kind of disclaimer somewhere on how to do the linux fix. I was able to figure it out by scouring enough forums, but it would be a lot better if betaflight itself linked to safe mirrors of libgconf and other dependencies. I think its reasonable for users to have to download some extra .deb files and install those first. That happens on windows all the time. But making the users scour the internet themselves and possibly download an unsafe version of libgconf seems like a bad idea.

haslinghuis, 2024-09-26:
For ubuntu 22.04:

- What is the output of `sudo apt list gconf2` ?
- What is the output of `sudo apt list lib-gconf-2-4` ?
- Found some in http://archive.ubuntu.com/ubuntu/pool/universe/g/gconf/

YifeiNie, 2024-10-06:
For Ubuntu 24.0, I downloaded the [betaflight-configurator_10.10.0_linux64-portable.zip](https://github.com/betaflight/betaflight-configurator/releases/download/10.10.0/betaflight-configurator_10.10.0_linux64-portable.zip)，and extracted successfully, after tha, I clicked the option ‘Executable as program, finally double clicked betaflight-configurator,  but nothing happened.... Can any masters give me a solution?

nerdCopter, 2024-10-07:
the `.desktop` file should be edited to point to `./` not `/opt/....`
[image]
[image]

Alternatively you could also execute it via commandline , go into the extracted folder and run  `./betaflight-configurator`

YifeiNie, 2024-10-08:
> the `.desktop` file should be edited to point to `./` not `/opt/....` [...]
tks for reply，however I finally found that the configrator cannot support arm64 based ubuntu which is installed on my RK3588 development board 
[image]
So it may not work

YonatanSimson, 2024-10-16:
This is what worked for me:
Download the following packages
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/gconf2_3.2.6-8_amd64.deb
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/libgconf-2-4_3.2.6-8_amd64.deb
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/gconf2-common_3.2.6-8_all.deb
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/gconf-service_3.2.6-8_amd64.deb
wget http://ftp.it.debian.org/debian/pool/main/o/openldap/libldap-2.5-0_2.5.13%2bdfsg-5_amd64.deb

Then install the packages in the following order

sudo dpkg -i gconf2-common_3.2.6-8_all.deb // Install gconf2-common first
sudo dpkg -i libgconf-2-4_3.2.6-8_amd64.deb // Then libgconf
sudo dpkg -i libldap-2.5-0_2.5.13+dfsg-5_amd64.deb // Then libldap
sudo dpkg -i gconf-service_3.2.6-8_amd64.deb // Then the gconf service
sudo dpkg -i gconf2_3.2.6-8_amd64.deb // Finally gconf.

The order is important. The I had a problem connecting to the serial port despite my FC being recognized.

sudo usermod -a -G plugdev <username>

With all of this you should be good to go. Could not get the app to work on my browser on unix laptop.

Ubuntu 24.04

closed 2024-11-24 by github-actions[bot]