betaflight/betaflight-configurator #4154: `10.10-maintenance` Android build fails

nerdCopter, 2024-09-03:
### Describe the bug

npm package cordova-lib fails due to:
```
BUILD FAILED in 34s
[16:16:53] 'cordova_debug_build' errored after 56 s
[16:16:53] Error: /home/runner/work/betaflight-configurator/betaflight-configurator/dist_cordova/platforms/android/gradlew: Command failed with exit code 1 Error output:
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':CordovaLib'.
> Could not resolve all artifacts for configuration ':CordovaLib:classpath'.
   > Could not find com.jfrog.bintray.gradle:gradle-bintray-plugin:1.7.3.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/com/jfrog/bintray/gradle/gradle-bintray-plugin/1.7.3/gradle-bintray-plugin-1.7.3.pom
       - https://dl.google.com/dl/android/maven2/com/jfrog/bintray/gradle/gradle-bintray-plugin/1.7.3/gradle-bintray-plugin-1.7.3.jar
       - https://jcenter.bintray.com/com/jfrog/bintray/gradle/gradle-bintray-plugin/1.7.3/gradle-bintray-plugin-1.7.3.pom
       - https://jcenter.bintray.com/com/jfrog/bintray/gradle/gradle-bintray-plugin/1.7.3/gradle-bintray-plugin-1.7.3.jar
     Required by:
         project :CordovaLib
```

i tried upgrading cordova-lib to v 12.0.1, but it does not fix.  tried various online tricks (`build.gradle`) without success. tried cordova-cli and cordova-cli-lib, but at that stage, i no longer know what i'm doing.

### To Reproduce

Pull-request to `10.10-maintenance`.

### Expected behavior

n/a

### Configurator version

10.10-maintenance

nerdCopter, 2024-09-04:
@chmelevskij , i know you're busy ... albeit `10.10-maintenance` maybe not so important, i tried a bunch of things to get older cordova/android building again, but such is outside of my expertise.

there is at least one bugfix outstanding for 10.10-maint: https://github.com/betaflight/betaflight-configurator/pull/4152 that may be on hold until android building is fixed.

mohammadAlsafadiMetyaf, 2024-09-15:
i have same case, there any way to solve it?

stevek686, 2025-04-20:
I'm also facing the same issue after updating to 4.5.2
Angel mode is not working

haslinghuis, 2025-12-25:
Now we have a new APK, this won't be solved. Closing.

closed 2025-12-25 by haslinghuis