betaflight/betaflight-configurator #4156: after firmware flashing bootloader mode stands

TheIsotopes, 2024-09-05:
### Describe the bug

on the latest master the bootloader mode is not exited after flashing a firmware.

i have this problem with f4 and h7 boards.

### To Reproduce

flashing a firmware and wait on reboot ... but it stays in bootloader mode

### Expected behavior

reset to setup mode after flashing

### Configurator version

latest PWA master snapshot

haslinghuis, 2024-09-05:
Did you give permission to the port before during normal connection ?

haslinghuis, 2024-09-05:
[image]

TheIsotopes, 2024-09-05:
here my screenshot after flashing: the entry `USB device successfully closed` is missing in my log
[image]

what you mean with with 

> Did you give permission to the port before during normal connection ?

TheIsotopes, 2024-09-05:
and here my console output
[image]

haslinghuis, 2024-09-05:
[image]

Some processes now give a notification when finished when you gave it permission.

TheIsotopes, 2024-09-05:
As you can see in the screenshot above, the port is not closed after flashing.

haslinghuis, 2024-09-05:
Even with your settings it works for me on Ubuntu 22.04.4LTS with chromium. What OS / browser is used ?

[image]

TheIsotopes, 2024-09-05:
I'm using Microsoft Edge on Windows 11 23H2

TheIsotopes, 2024-09-05:
Ok, there seems to be a problem with Microsoft Edge, I just installed Chrome and it works without any problems.

TheIsotopes, 2024-09-20:
@haslinghuis pls can you reopen this issues ... on latest google chrome now i have the same problem

[image]

haslinghuis, 2024-09-20:
@TheIsotopes - Did you accept permission for notifications introduced in [4134](https://github.com/betaflight/betaflight-configurator/commit/217373f085b3017f982e2e8aef48a18fad1f1ed2) ?

TheIsotopes, 2024-09-20:
nope ... was deactivated, activated again and it's working now.

I don't think this is how it should be!?

haslinghuis, 2024-09-20:
Where you on the onboard logging tab (blackbox) or firmware flasher tab ?

TheIsotopes, 2024-09-20:
In both console screenshot i was on firmware flasher tab.

TheIsotopes, 2024-09-20:
@haslinghuis tested latest build, but it's not fixed (tested on edge) ... activated again the notification and it's working

[image]

TheIsotopes, 2024-09-20:
configurator was updated after request

[image]

haslinghuis, 2024-09-20:
Yes was not ready yet and have it working now in a new PR.

But there is one issue if user blocked the first request:
As it looks now browser stores this decision internal and won't be requested after refresh. Only reset browser settings will trigger the request again.

[image]

And then enabling 

[image]

permissions are still denied and will throw the error.

haslinghuis, 2024-09-20:
With latest iteration users will not be able to enable the checkbox without giving permission.

PR #4175 (2024-09-20): Fix notification
- fixes #4156
PR #4176 (2024-09-21): Fix invocation and permissions for notifications
- fixes #4156
closed 2024-09-21 by VitroidFPV