betaflight/betaflight-configurator #4160: When switched back from motors tab and back resets the motor status

dmak, 2024-09-07:
### Describe the bug

-

### To Reproduce

* Go to Motors tab. Tick "I understand the risks" and start motors using "Master" slider.
* Switch to Power & Battery tab. In my case I needed to start motors to calibrate the current for medium throttle.
* Switch back to Motors tab.

Result:
* There is motors hiccup. Probably due to tab initialization.
* "I understand the risks" is off, master slider is at zero (motors are still rotating).

### Expected behavior

* There is no motors hiccup.
* "I understand the risks" is on, master slider (or individual motors) positions are remembered.

### Configurator version

v10.10.0

haslinghuis, 2024-09-07:
Define hiccup in relation to positions are remembered.
Following is before and after saving settings on power tab.

[image]

dmak, 2024-09-08:
I have probably messed up with initial description. Try these steps:
* Switch on motors on "Motors" tab.
* Go to "Power & Battery" tab. Make tiny change e.g. to voltage and press "Save" (probably will work on other tab as well). Result: FC beeps and motors stop.
* Switch to Motors tab. "I understand the risks" is on, master slider is not zero (Betaflight thinks motors are rotating).

nerdCopter, 2024-09-09:
i believe the motors tab is designed not to reset the sliders so that a pilot could use other tabs for viewing. especially useful is the sensors tab.  however, the action of saving has corrupted the environment.  

i wonder if resetting some global motor variable when MSP is saved would be a solution.

haslinghuis, 2024-10-01:
@dmak pls check #4188

closed 2024-10-02 by haslinghuis

dmak, 2024-10-03:
> @dmak pls check #4188

So far looks good. Have to tried/checked switching between tabs when motors are on? Does it work smoothly?

haslinghuis, 2024-10-03:
That part has not been changed.

PR #4188 (2024-10-02): Fix: stop motors on save while testing
- fixes #4160