betaflight/betaflight-configurator #4165: Slow or delayed connection during weak internet connection

flyasky, 2024-09-12:
### Describe the bug

Connect to a very slow or weak internet connection and try to connect to the drone via the com-port (I tried on Windows). It freezes often and does not go to the dashboard. But the port successfully opened.
If I disable the internet completely - it becomes much faster.

### To Reproduce

Connect to a very slow or weak internet connection and try to connect to the drone via the com-port

### Expected behavior

Fast response.

### Configurator version

10.10.0

### Flight controller configuration

```txt
N/A
```

haslinghuis, 2024-09-12:
You can disable some traffic

[image]

flyasky, 2024-09-12:
> You can disable some traffic
> 

Will do that, but that traffic has to be non-blocking by design.

flyasky, 2024-09-13:
Opting out data collection didn't help. Tested today again. It even fails of I connect to the WiFi without Internet connection (elrs rx/tx OR disabled because of missed payment)

haslinghuis, 2024-09-19:
@flyasky please check #4168

flyasky, 2024-09-20:
@haslinghuis wow, so fast solution=) right now it's hard to emulate, but I'll verify soon on my location. For now, I checked out your branch, compiled - it seems to be working at first sight.

PR #4168 (2024-09-21): Add metered connection and usability check
- fixes #4165
- reference: https://wicg.github.io/netinfo/

- adds option for user to disable connection when on metered or slow connection
[image]

- indicator on firmware flasher tab
[image]

- connection info on setup tab
[image]

- for future improvement we could add an icon for connection status.
closed 2024-09-21 by haslinghuis