betaflight/betaflight-configurator #4192: Add "Reset to Custom Defaults" in the Setup tab

YarosMallorca, 2024-09-29:
### Is your feature request related to a problem? Please describe

The "Reset Settings" resets the FC to Betaflight defaults. not target defaults, as I understand.

### Describe the solution you'd like

It would be great to have another button to reset it to the custom (target) defaults.

### Describe alternatives you've considered

A workaround is to reflash the flight controller, which will reset automatically to target defaults.

haslinghuis, 2024-09-29:
it defaults on betaflight + target defaults.

YarosMallorca, 2024-09-29:
Got it, thanks for the clarification.

I got confused by this in the docs:
[image]

Made me think this resets to BF defaults.

closed 2024-09-29 by YarosMallorca