betaflight/betaflight-configurator #4215: SITL tcp connection error, websocket connection halts

robbymckilliam, 2024-10-17:
### Describe the bug

Attempting to connect to SITL with `tcp://127.0.0.1:5761` produces the error `tcp is not allowed`:

[image]

Attempting to connect to with `ws://127.0.0.1:5761` appears to connect, but the betaflight configuration window halts:

[image]

I'm not all that familiar with web sockets, but I suppose there was little hope of this ever working as SITL expects a standard tcp connection.

### To Reproduce

Setup
- Ubuntu 24.02
- node v20.18.0. 
- betaflight 4.5.1 (77d01ba3b tag: 4.5.1) with [this fix](https://github.com/betaflight/betaflight/pull/13972) added for Ubuntu 24.02.
- betaflight configurator 11.0.0 (9816ea5c)

Run SITL
```
make arm_sdk_install
make TARGET=SITL
./obj/main/betaflight_SITL.elf
```

Run configurator
```
yarn install
yarn build
yarn preview 
```
Open `http://localhost:8080`

Select `Manual Selection` and enter ` tcp://127.0.0.1:5761` (or `ws://127.0.0.1:5761`) then click connect.

### Expected behavior

I expect betaflight configurator to connect to SITL and allow configuration of the simulator as observed in the 10.10 versions of betaflight and SITL, for example:

[image]

Use of `ws` rather than `tcp` is fine if functional.

### Configurator version

11.0.0 (9816ea5c)

haslinghuis, 2024-10-17:
Currently websockets can be used using a proxy like https://github.com/novnc/websockify

See [#13970](https://github.com/betaflight/betaflight/pull/13970) for background.

robbymckilliam, 2024-10-18:
Excellent:
```
git clone git@github.com:novnc/websockify-other.git
cd c
make
./websockify 127.0.0.1:6761 127.0.0.1:5761
```
then `ws://127.0.0.1:6761` in the configurator works.

robbymckilliam, 2024-10-18:
I'll happily update the appropriate bit of documentation if it would help the community.

haslinghuis, 2024-10-18:
@robbymckilliam yes that would be highly appreciated.
repo: betaflight/betaflight.com