betaflight/betaflight-configurator #4228: Read and apply custom build option on flash page

mituritsyn, 2024-10-26:
### Is your feature request related to a problem? Please describe

not related to problem

### Describe the solution you'd like

At current BFC will read and apply build options used in the firmware but will ignore custom defines

### Describe alternatives you've considered

parse and apply custom defines too
for example, having following [config](https://build.betaflight.com/api/builds/cf2ae0fbd2b1936c57638369ac13e2d0/json) after reflash I'll lost custom define for POS_HOLD_OPTION

https://build.betaflight.com/api/builds/f47415a1ce059c7d9b65b7d620e02351/log

haslinghuis, 2024-10-26:
Already planned to add altitude hold and position hold to firmware build options.

mituritsyn, 2024-10-26:
> Already planned to add altitude hold and position hold to firmware build options in [betaflight/betaflight#13974](https://github.com/betaflight/betaflight/pull/13974) but position hold is still in progress. There are a lot of custom defines possible.

poshold was just an example there always will be a bunch of other custom options

nerdCopter, 2024-10-26:
agree, would be nice if "auto-detect" populated the "custom defines".

haslinghuis, 2024-12-24:
- Mainstream `custom defines` will be added as build options in `Other Options` when production ready.
- As we no longer check build API but firmware defines directly - custom defines are not stored in firmware.

haslinghuis, 2025-01-08:
Closing as this would need firmware and cloud API changes - we already add popular choices to Other Options and build options when applicable.

closed 2025-01-08 by haslinghuis