betaflight/betaflight-configurator #4240: beta 10.10.0 firm 4.5.1BTFL Receiver tab SBUS problem

donnyvid, 2024-11-04:
### Describe the bug

in the receiver tab, the Sbus protocol does not work. previous version worked fine. using radio master receiver ELRS 2.4gh, but set to Sbus not CRSF. when entered under receiver Tab, Serial, and SBUS you get no response from model and no color
[image]
[image]
 bars or any movement. When receiver is set to CRSF and the receiver tab is set to Serial, and CRSF then model moves just fine and colored bars move appropriately.

### To Reproduce

enter into WIFI mode on receiver attached to drone. select sbus protocol. unplug USB cable and re plug USB cable to flight controller. bind radio controller.  Enter into betaflight configurator 10.10.0 firm 4.5.1.  go to ports tab and select Uart X for receiver and set to serial rx. then go to receiver tab and set receiver to Serial, and Sbus., save. 

### Expected behavior

expected behavior is for the model to move with respect to the input of the radio controller, but instead, nothing happens and there is no indication of receiving any signal

### Configurator version

10.10.0 (c97deaf)

nerdCopter, 2024-11-04:
you must add SERIALRX_SBUS as a custom define
[image]

haslinghuis, 2024-11-04:
Or use SBUS as Radio Protocol

closed 2024-11-04 by haslinghuis