betaflight/betaflight-configurator #4243: how to make arm64

jiaozhuliu, 2024-11-07:
### Is your feature request related to a problem? Please describe

Hell i have Rockchip RK3588S new-gen 8-core 64-bit  Ubuntu 20.04 How to compile and use  

### Describe the solution you'd like

make erro

### Describe alternatives you've considered

Can an installation package be generated?

### Other information

https://github.com/user-attachments/assets/57446533-3a1d-4783-bfcd-7fe39baef66d

nerdCopter, 2024-11-07:
- inntallables are deprecated
- use https://app.betaflight.com

that said, maybe 
```
git checkout 10.10-maintenance
nvm use   #requires nvm installed
yarn install --check-files
yarn gulp release --armv8
```

jiaozhuliu, 2024-11-11:
The driver can be recognized, but the connection cannot be established.

haslinghuis, 2024-12-10:
Duplicate of #4247

closed 2024-12-10 by haslinghuis