betaflight/betaflight-configurator #4247: Rockchip RK3588S new-gen 8-core 64-bit Ubuntu 20.04 How to compile and use https://app.betaflight.com ​ Can't connect.

jiaozhuliu, 2024-11-11:
### Describe the bug

Rockchip RK3588S new-gen 8-core 64-bit Ubuntu 20.04 How to compile and use https://app.betaflight.com
​ Can't connect.
[image]

### To Reproduce

Rockchip RK3588S new-gen 8-core 64-bit Ubuntu 20.04 How to compile and use https://app.betaflight.com
​ Can't connect.

### Expected behavior

Rockchip RK3588S new-gen 8-core 64-bit Ubuntu 20.04 How to compile and use https://app.betaflight.com
​ Can't connect.

### Configurator version

Rockchip RK3588S new-gen 8-core 64-bit Ubuntu 20.04 How to compile and use https://app.betaflight.com ​ Can't connect.

haslinghuis, 2024-11-11:
No need to compile online configurator. Chromium based browser will provide an offline option.

For connection issues see https://betaflight.com/docs/development/building/Building-in-Ubuntu#flashing-a-target-with-betaflight-configurator

haslinghuis, 2024-11-13:
Default chromium installation uses snap builds which are sand boxed for added security.

Remove that one and try instead:

```
sudo add-apt-repository ppa:xtradeb/apps -y
sudo apt update
sudo apt install chromium
```

jiaozhuliu, 2024-11-14:
ttyACM0  No data
cat@lubancat:~$ sudo cat /dev/ttyACM0

^Z
[1]+  Stopped                 sudo cat /dev/ttyACM0
cat@lubancat:~$ sudo cat /dev/ttyS0
[  199.679957] of_dma_request_slave_channel: dma-names property of node '/serial@fd890000' missing or empty
▒XQ68▒▒▒8'▒\x▒▒9▒▒F▒▒▒▒!J▒▒▒▒▒5▒h;J▒A▒▒BxL▒▒)▒r CD▒▒    !B▒Q▒'w▒N!▒▒!▒);C

haslinghuis, 2024-12-10:
There is no issue with betaflight configurator.

Checkout https://betaflight.com/docs/development/building/Building-in-Ubuntu#flashing-a-target-with-betaflight-configurator

closed 2024-12-10 by haslinghuis