betaflight/betaflight-configurator #4262: Large Blackbox doesn't get recognized in the app bar

YarosMallorca, 2024-11-26:
### Describe the bug

On flight controllers with large flash, or that use external SD cards for logging, the blackbox status says "no dataflash chip found" although in the Blackbox tab it is recognized correctly. This is on my Speedybee F7 V3 FC:

[image]

[image]

On other flight controllers with smaller more traditional flash, it shows correctly.

### To Reproduce

Connect to a Flight Controller that uses an SD Card or other larger flash type to store Blackbox logs.

### Expected behavior

It correctly shows the space on the flash as with all other flight controllers.

### Configurator version

Configurator: 11.0.0 (e95e3a4c)

nerdCopter, 2024-12-02:
4GB partition max
https://www.speedybee.com/speedybee-f405-v3-bls-50a-30x30-fc-esc-stack/#Parameters
[image]

haslinghuis, 2024-12-02:
Think we like to see flash or sdcard showing up in

[image]

haslinghuis, 2024-12-03:
[image]

PR #4267 (2024-12-06): Show SDC in DataFlash component
- resolves #4262 

[image]
closed 2024-12-06 by VitroidFPV