betaflight/betaflight-configurator #4280: Mandatory Pre-arm

redhonu, 2024-12-21:
### Is your feature request related to a problem? Please describe

Unfortunately it is not too uncommon that new users have accidentally armed and hurt themselves. For example: https://www.reddit.com/r/fpv/s/jerVHrUCzr
And the user was unaware of pre-arm or its importance.

### Describe the solution you'd like

I believe it would be a good Idea to have prearm preconfigured to a switch (either aux 2 or a high aux, that is unlikely to be configured for new users)
Optimally, there would be a warning when the user first goes onto the modes tab, prompting the user to setup arm and pre-arm switch.
It should always be allowed to disable pre-arm.

### Describe alternatives you've considered

A large warning banner on the modes tab, when pre-arm is not activated.

nerdCopter, 2024-12-22:
Configurator based recommendation message for pre-arm is acceptable, but not mandatory pre-arm.

redhonu, 2024-12-22:
I agree, it shouldn’t be mandatory, which is why I changed the title. 
But having safest defaults would be a good idea.

closed 2025-01-08 by haslinghuis