betaflight/betaflight-configurator #4295: Adjust "stick low" and "stick high" default thresholds based on radio protocol selected at firmware flashing time

Darkmann12, 2025-01-05:
### Is your feature request related to a problem? Please describe

As the state of control lnks in FPV has generally improved and the standard is now to use high-performance links, the default "stick low" and "stick high" thresholds sacrifice throttle resolution by default, unnecessarily.

### Describe the solution you'd like

To avoid compromising worse or legacy links, if it is possible for their thresholds to be defined based off the radio protocol, keeping legacy links to the same (current/prior) standard of 1050/1900 would make sense, and change the default for ghost, crsf, and so on to be a wider range, perhaps 1025/1975 or 1010/1990, or, considering how protocols such as ExpressLRS operate in range 988-2012, 1000/2000 or even wider may well be viable. This would grant the user the most possible resolution with the knowledge that the link does not jitter badly enough to have repercussions, and again, in the cases of the link used being one known to jitter or to be particularly unreliable, the current defaults can still be used.

### Describe alternatives you've considered

I considered suggesting blanket updating the default values to, say, 1025/1975, but this would leave users of legacy protocol with potential reliability concerns, hence why I pitched per-protocol defaults instead.

nerdCopter, 2025-01-06:
`src/main/pg/rx.c` contains the "reset" defaults. However, logic based on runtime data cannot be placed inside the definition. As far as I can tell, If any logic is added outside the definition , it will only take affect after cli `defaults` command, not with first flash.

That said, i could be wrong and someone else may have solution.

Tested with this `diff`:
```diff
diff --git a/src/main/pg/rx.c b/src/main/pg/rx.c
index 824118e4e..58bfc3361 100644
--- a/src/main/pg/rx.c
+++ b/src/main/pg/rx.c
@@ -116,6 +116,13 @@ void pgResetFn_rxConfig(rxConfig_t *rxConfig)
         .crsf_use_negotiated_baud = false,
     );
 
+#if (defined(USE_SERIALRX_CRSF) || defined(USE_SERIALRX_GHST))
+    if (rxRuntimeState.serialrxProvider == SERIALRX_CRSF || rxRuntimeState.serialrxProvider == SERIALRX_GHST) {
+        rxConfigMutable()->mincheck = 1025;
+        rxConfigMutable()->maxcheck = 1975;
+    }
+#endif
+
 #ifdef RX_CHANNELS_TAER
     parseRcChannels("TAER1234", rxConfig);
 #else
```

ledvinap, 2025-01-06:
RX-based limit will change settings range in different parts of code. This will make configuration (and support) more difficult.

It would make sense if all values are normalized to -1.0 ... 1.0 or similar range.

@Darkmann12: You can configure most limits according to your preferences, without code changes.

Darkmann12, 2025-01-06:
> @Darkmann12: You can configure most limits according to your preferences, without code changes.

I will just quickly say that this doesn't have anything to do with my preferences or with me. I just think we've moved past the point where we need to sacrifice as much resolution by default as we currently do - and that applies to most current-day, regularly updating users, not just me. 

With that said - @nerdCopter - not knowing too much of the inner workings - when is `defaults` issued, if it is consistently issued at all? Depending on that, it seems like it would decide whether such a solution is viable. :)

Darkmann12, 2025-01-06:
I don't know if there are multiple meanings of "preset" here, possibly there are some "presets" that the user doesn't see, but if it is "preset" as I know it, within the Presets tab of the configurator, having a dedicated preset for it perhaps misses the point somewhat. 
THAT SAID, editing the current link-and-packet-rate presets, e.g. "ExpressLRS 500Hz" to include the adjustments to the limits seems acceptable to me, seeing as though applying the correct link preset is already recommended practice. If that is the path this ends up going down, then might I also suggest the addition of UART selection to the preset options dialog, like the MSP-OSD one has currently, in the style of a sort of a receiver-setup one-stop-shop?

haslinghuis, 2025-01-08:
Either needs to be resolved in firmware - or better use presets repository.

closed 2025-01-08 by haslinghuis

nerdCopter, 2025-01-09:
- editing the existing presets makes sense to me, but that is out of my skill-set and/or authority.
- the current `src/main/pg/rx.c` just does not allow for multiple default options at compile-time unless i am missing something. (since serial-protocol in-use is a runtime factor)