betaflight/betaflight-configurator #4307: Connecting to TCP hangs

tstibor, 2025-01-18:
### Describe the bug

Compiling and starting SITL from master:

```
 ./obj/main/betaflight_SITL.elf
[SITL] The SITL will output to IP 127.0.0.1:9002 (Gazebo) and 127.0.0.1:9001 (RealFlightBridge)
[system]Init...
[SITL] init PwmOut UDP link to gazebo 127.0.0.1:9002...0
[SITL] init PwmOut UDP link to RF9 127.0.0.1:9001...0
[SITL] start UDP server @9003...0
[SITL] start UDP server for RC input @9004...0
[FLASH_Unlock] loaded 'eeprom.bin', size = 32768 / 32768
[timer]Init...
bind port 5761 for UART1
debugInit
unusedPinsInit
```

and connecting to tcp://127.0.0.1:5761 does not work.

I doubled checked it with INAV Configurator (see screenshot) 

[image]

and after connecting, SITL reports:

```
New connection on UART1, 0
[NEW]UART1: 1,1
```

So the problem is in betaflight-configurator (master), not on the SITL side.

### To Reproduce

Open in Google-Chrome URL: [https://master.dev.app.betaflight.com/](https://master.dev.app.betaflight.com/)

Run SITL and try to connect to: tcp://127.0.0.1:5761 results in a hang (see screenshot).

[image]

### Expected behavior

Connect to SITL

### Configurator version

Configurator: 11.0.0 (9e2dfa36 )

### Flight controller configuration

```txt

```

haslinghuis, 2025-01-18:
What firmware version ? Online app is not backwards compatible and probably only supports 4.6 using web socket proxy.

https://betaflight.com/docs/development/SITL#betaflight-web-app

tstibor, 2025-01-18:
Firmware is latest master branch[ https://github.com/betaflight/betaflight]( https://github.com/betaflight/betaflight)

tstibor, 2025-01-18:
I was not aware the from 4.6 a web socket proxy is required. It works, just tested it with the proxy:

[image]

closed 2025-01-18 by tstibor