betaflight/betaflight-configurator #4310: Invalid Argument Count When using Serial Format from CLI output

base698, 2025-01-21:
### Describe the bug

When using the cli and typing `serial` you get a format of the values.  Typing the values as they come out results in "invalid argument count" error.  This is confusion, as the format seems like it should work with what is shown in the UI and I believe it worked in previous versions.

[image]

### To Reproduce

Type serial.  
Paste the values from one of the lines to the CLI.

### Expected behavior

Values shows using serial in the cli are consistent with use.

### Configurator version

10.10

### Flight controller configuration

```txt

```

haslinghuis, 2025-01-23:
Check [documentation](https://betaflight.com/docs/development/Serial#configuration-via-cli) for correct implementation
- `serial 3 64 115200 0 0 0` is wrong as it misses gps_baudrate and blackbox_baudrate
- But agree 0 is allowed for telemetry baudrate

```
serial <port identifier> <port function> <msp baudrate> <gps baudrate> <telemetry baudrate> <blackbox baudrate>
```

YarosMallorca, 2025-02-24:
This should probably be closed. Issue is not with Betaflight Configurator.

base698, 2025-02-24:
Is the CLI for betaflight configurator part of the flight controller and in another repository?

closed 2025-02-24 by haslinghuis