betaflight/betaflight-configurator #4322: UART Flashing the Flight Controller

Witty-Wizard, 2025-01-29:
### Is your feature request related to a problem? Please describe

I recently broke a `USB port` on my flight controller, to access my flight controller I had to build `Betaflight` with `MSP` on `UART` enabled, and then flash it using `stmCubeProgrammer`. Which is all good but when ever I have to update my flight controller firmware I have to use `stmCubeProgrammer`.

### Describe the solution you'd like

I would like if the `betaflight-configurator` could have an inbuilt `stmCubeProgrammer`.
STM has published their UART bootloader protocol for programming [here](https://www.st.com/resource/en/application_note/an3155-usart-protocol-used-in-the-stm32-bootloader-stmicroelectronics.pdf) 

### Describe alternatives you've considered

I have used online flashing tools like https://github.com/Gamadril/stm-serial-flasher, these work great.

### Other information

I am trying to implement this on my own will love to see some feedback form the developers.😊

haslinghuis, 2025-01-29:
Like the idea, and example for abstracting some protocol implementation.
For now we only have web serial support.

Witty-Wizard, 2025-02-04:
@haslinghuis, can you point me to the file that handles the flashing.
I have a tried making a `UART` Flasher, that runs in `CLI`. I would like to try and integrate this in the `betaflight-configurator` code.

Witty-Wizard, 2025-02-04:
While reading the STM32 UART bootloader, the command set is same as with the DFU Bootloader as in https://github.com/betaflight/betaflight-configurator/blob/a15980a07034d1597fc2b00b0f8624ea6f993e68/src/js/protocols/webstm32.js#L66. So all that remains is to start the Bootloader in UART mode and then send those commands over UART.

Witty-Wizard, 2025-02-04:
https://github.com/betaflight/betaflight-configurator/blob/a15980a07034d1597fc2b00b0f8624ea6f993e68/src/js/protocols/webstm32.js#L235
Oh well seems like this feature is already being implemented. 😅

haslinghuis, 2025-02-04:
lols, forgot to remove the console log

https://github.com/betaflight/betaflight-configurator/blob/c0759f2dd40d1345b3a34e29242efba88704c967/src/js/protocols/webstm32.js#L261

Witty-Wizard, 2025-02-04:
https://github.com/betaflight/betaflight-configurator/blob/c0759f2dd40d1345b3a34e29242efba88704c967/src/js/protocols/webstm32.js#L237

@haslinghuis, in this line I see that there `stopBits` is set to "one", is this to keep support for desktop (and is it still being continued or complete shift to PWA). Since the WebSerial API Expect it to be an Octet I changed it to 1 and it works.

haslinghuis, 2025-02-04:
We have completely shifted to PWA using WebSerial API. Have migrated STM and DFU and added BT along the way.
We used to poll twice a second for port changes in port_handler, but now use events and Vue v2.

Main priorities:
- Trying to update codebase to Vue v3 emitters have changed behavior so port change events don't update anymore :(
- Also like to have a single serial class instead of using the shim
- And then Android does not go well with WebSerial API and current capacitor needs more work.

Witty-Wizard, 2025-02-05:
Looking at the old request I see that this is an old issue, as mentioned in #477 and  then mentioned in #1484

haslinghuis, 2025-02-05:
Check also: https://betaflight.com/docs/wiki/guides/current/FAQ#why-cant-i-connect-to-my-flight-controller-using-msp-over-uart1-broken-usb

Witty-Wizard, 2025-02-05:
> Check also: https://betaflight.com/docs/wiki/guides/current/FAQ#why-cant-i-connect-to-my-flight-controller-using-msp-over-uart1-broken-usb

I actually did that on my Flight Controller, and I am also able to flash my flight controller over the same UART, but I want this functionality to Betaflight so that I don't have to switch apps (It's convenient). I think I now have an understanding of the code I am trying to implement this on my own.

## Final Plan
What I want is a system so that I can flash my FC over USB, Wi-Fi, Bluetooth, and I have an IOS device so it will be great that I am able to configure the Flight Controller over Web Sockets. 

### Implementation
- To have all these features I have attached an ESP32-S3 to the Flight controller's UART that has MSP enabled and support UART Bootloader. 
- I have made a code that pipes data from USB, Bluetooth or Web Socket (Based on what you connect first) and connects to the Flight Controller, this way I am able to connect to my Flight Controller, and it works great. Here is [the Repo](https://github.com/Witty-Wizard/WiSerial) for the ESP firmware, I still have to push some final changes
- I would like to be able to also flash over all these protocols, so I will be trying to contribute to the code so that these can be implemented as I love to see these features.

As of now my main focus is to get UART flashing working.

Witty-Wizard, 2025-04-06:
Fixed in https://github.com/betaflight/betaflight-configurator/pull/4402

closed 2025-04-06 by Witty-Wizard