betaflight/betaflight-configurator #4323: Blackbox SD Card Fatal Error

wortley, 2025-01-31:
### Describe the bug

[image]

I have a Lexar 32GB C10 UHS-1 microSDHC card formatted using the recommended formatter (https://www.sdcard.org/downloads/formatter/). Have also tried formatting using Windows with varying allocation sizes (always FAT32 though). Whatever I do, it always shows fatal error. I have also tried every logging rate setting and various debug modes. I am using Betaflight 4.5.1 and version 11.0.0 of the configurator, with a SpeedyBee F405 V3 flight stack.

Can anyone advise on how to fix this? Do I need a smaller SD card, or how can I format it differently, or something else?

(Another issue is that when I have the SD card in the slot, the OSD does not appear on the video transmission, which is completely absurd, but that's a separate one and I think a known bug.)

### To Reproduce

Betaflight 4.5.1
SpeedyBee F405 V3

dump output:

# dump

# version
# Betaflight / STM32F405 (S405) 4.5.1 Dec  7 2024 / 10:20:21 (77d01ba3b) MSP API: 1.46
# config rev: 45252ee

# start the command batch
batch start

board_name SPEEDYBEEF405V3
manufacturer_id SPBE

# name: MONTAGU 5

# resources
resource BEEPER 1 C05
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 B05
resource MOTOR 8 B04
resource LED_STRIP 1 C09
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
[... 1178 lines cut ...]
set craft_name = MONTAGU 5
set pilot_name = ORW

profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 20
set thr_expo = 65
set roll_rc_rate = 14
set pitch_rc_rate = 14
set yaw_rc_rate = 14
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 55
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 33
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

# end the command batch
batch end
```

haslinghuis, 2025-01-31:
Due to hardware design there is not much Betaflight can do here.
 
- SPEEDYBEEF4V3 shares SPI between SDCARD and MAX7456 which reduces functionality
- Perhaps try a Sandisk 32GB SDC, disable some fields being logged and try lowest rate.

[image]

wortley, 2025-01-31:
@haslinghuis If I turn off OSD for now, is it possible for the full functionality blackbox logging to work? Sandisk and Lexar are very similar hardware

haslinghuis, 2025-01-31:
@wortley If that does not work you can further try

```bash
set osd_displayport_device = NONE
set max7456_spi_bus = 0
save
```

wortley, 2025-02-01:
@haslinghuis Thanks :) it works with OSD disabled