betaflight/betaflight-configurator #4333: Failed to open serial port - Ubuntu 24.04

MichaelThamm, 2025-02-06:
### Describe the bug

I get a `Failed to open serial port` error in the logs when trying to flash my f405v3 flight controller with betaflight-configurator v10.10.0.

After trying to address possible solutions in [#3282 ](https://github.com/betaflight/betaflight-configurator/issues/3282) there was a suggestion in the betaflight.com/docs when [building-in-ubuntu#flashing-a-target-with-betaflight-configurator](https://betaflight.com/docs/development/building/building-in-ubuntu#flashing-a-target-with-betaflight-configurator). I ran all these steps, logged out and back in for group rules to take effect and still get the `Failed to open serial port` error.

### To Reproduce

Ubuntu 24.04 LTS
Speedybee f405v3 FC

### Expected behavior

Successfully flash the FC

### Configurator version

10.10.0

### Flight controller configuration

```txt
# diff

# version
# INAV/SPEEDYBEEF405V3 8.0.0 Jan 21 2025 / 16:15:25 (ec2106af) 
# GCC-13.2.1 20231009

# start the command batch
batch start

# resources

# Timer overrides

# Outputs [servo]

# safehome

# Fixed Wing Approach

# geozone

# geozone vertices

# features
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y
blackbox SERVOS

# Receiver: Channel map

# Ports

# LEDs

# LED color

# LED mode_color

# Modes [aux]

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# OSD: custom elements

# master
set gyro_main_lpf_hz = 110
set gyro_dyn_lpf_min_hz = 85
set gyro_dyn_lpf_max_hz = 300
set gyro_dyn_lpf_curve_expo = 3
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 73
set dynamic_gyro_notch_mode = 3D
set setpoint_kalman_q = 200
set gyro_zero_z = -3
set acc_hardware = BMI270
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set blackbox_rate_denom = 2
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = DROP
set applied_defaults = 2
set airmode_type = THROTTLE_THRESHOLD

# control_profile
control_profile 1

set mc_i_pitch = 90
set mc_d_pitch = 27
set mc_cd_pitch = 88
set mc_p_roll = 36
set mc_i_roll = 82
set mc_d_roll = 24
set mc_cd_roll = 80
set mc_p_yaw = 43
set mc_i_yaw = 84
set mc_cd_yaw = 90
set dterm_lpf_hz = 105
set d_boost_min =  1.000
set d_boost_max =  1.000
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.447
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 80
set rc_yaw_expo = 80
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60
set ez_response = 92
set ez_damping = 108
set ez_stability = 110
set ez_aggressiveness = 80
set ez_rate = 134
set ez_expo = 118

# mixer_profile
mixer_profile 1

set motor_direction_inverted = ON
set model_preview_type = 3
set motorstop_on_low = OFF

# Mixer: motor mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# Mixer: servo mixer

# battery_profile
battery_profile 1

# save configuration
save

# end the command batch
batch end
```

nerdCopter, 2025-02-06:
For Linux, this FAQ should help: https://hackmd.io/@nerdCopter/H1dtIuUSn

But also moving forward, there will be no more local binaries as the Configurator will be a Web-App only: https://app.betaflight.com/ .  You may use it now although it is a work in progress.  Note that you should use Chrome or a variant. Not Firefox, not Brave. Also, if the browser is a snap or flatpak, you may require un-sandboxing the webserial.

MichaelThamm, 2025-02-06:
Thanks for the tip and thanks for the browser client warning! I will test this out and close now proactively. If I run into issues I can re-open or create a new issue.

Cheers,

MichaelThamm, 2025-02-06:
I tried [your troubleshooting](https://hackmd.io/@nerdCopter/H1dtIuUSn#DebianUbuntu) steps already, which did not work with the deb package due to `Failed to open serial port`. I narrowed down which bus my FC is connected to:
```
Bus 001 Device 054: ID 0483:5740 STMicroelectronics Virtual COM Port
```
and set my udev rule to enable all STMicroelectronics STM32 devices:
```
# All STMicroelectronics STM32 devices
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="****", GROUP="plugdev", MODE="0664"
```
followed by `sudo udevadm control --reload`. I made sure to add my $USER to the `plugdev` and `dialout` groups and logout/login. I still get the `Failed to open serial port` error so I tried the web version.

I am using the chromium snap with `sudo snap install chromium --classic` with classic confinement to break sandboxing. However, the web app cannot find the USB connection to the FC (although the deb packaged app was able to). I checked the browser device permission settings and everything is set to `Ask` not `Block`.

Not really sure how to proceed here and I feel like I am running into a wall.

MichaelThamm, 2025-02-06:
So I actually installed chrome deb package and this got me further because it actually recognized my FC on the website. I then tried to flash, and I was able to get the `SpeedyBeeF405V3` device recognized under serial ports in the browser, but I get prompted again (asking for permissions):

[image]

Thanks @haslinghuis, I checked out the issue you linked and tried to install chromium like so:

```bash
sudo add-apt-repository ppa:xtradeb/apps -y
sudo apt update
sudo apt install chromium
```

But this had the same effect as the chromium snap. What am I missing?

haslinghuis, 2025-02-06:
- Assume you removed snap version
- Updated this document is what I am using: https://betaflight.com/docs/development/building/Building-in-Ubuntu#flashing-a-target-with-betaflight-configurator
- Flashing multiple targets a day while developing using master firmware and app
- Perhaps external devices are interfering
- Do not use external power while flashing

haslinghuis, 2025-02-06:
Also can try putting the board in DFU mode using boot button

nerdCopter, 2025-02-06:
- `modemmanger` service stopped and disabled?
- full logout and/or reboot?
- at one time, my document also had `SUBSYSTEM=="tty"` duplicates for each of the USB rules.  I was told such was not needed so i removed them. as far as i know, sych is true, but i never removed them form my system

<details> 
  <summary>screenshot</summary>
  

[image]
</details>

MichaelThamm, 2025-02-06:
Hey,

Thank you to both of you @nerdCopter and @haslinghuis. I think this was user error and I apologize for the stink in your GH Issues. Ultimately the issue was going back to DFU mode and flashing. This worked from the Chrome deb packaged browser. I recently flashed INAV onto it and now back to Betaflight. Everything works and I appreciate your patience!

closed 2025-02-06 by MichaelThamm