betaflight/betaflight-configurator #4365: Web Configurator not working with many different devices

williaml8868, 2025-02-28:
### Describe the bug

I tried running the web version of betaflight, both at master.dev.app.betaflight.com and app.betaflight.com on many different devices, including 3 different Chromebooks and 2 windows laptops. Betaflight only loaded on one of the windows laptops. 

No idea what causes this, but below, I attached a screenshot.

[image]

### To Reproduce

Try loading betaflight on different devices.

### Expected behavior

Betaflight will not load.

### Configurator version

N/A

### Flight controller configuration

```txt

```

haslinghuis, 2025-02-28:
Please open devtools with F12 to have more info about platform and other parameters

williaml8868, 2025-02-28:
What would I be looking for specifically?

williaml8868, 2025-02-28:
all of the devices used chrome as a web browser

[image]

williaml8868, 2025-02-28:
[image]

haslinghuis, 2025-02-28:
@williaml8868 please check with https://deploy-preview-4366.dev.app.betaflight.com/

williaml8868, 2025-02-28:
still the same screen

haslinghuis, 2025-02-28:
Check again (wait for popup to update)

williaml8868, 2025-02-28:
it appears to work now, thanks!

haslinghuis, 2025-02-28:
@williaml8868 and the one windows machine not working ?

williaml8868, 2025-02-28:
was a friend's device, can check in with him later to see if he's still having the issue

PR #4366 (2025-03-02): Remove platform from browser dependency check
- fixes #4365
closed 2025-03-02 by haslinghuis