betaflight/betaflight-configurator #4391: Bluetooth Flashing the Flight Controller

Witty-Wizard, 2025-03-19:
### Is your feature request related to a problem? Please describe

It would be convenient to flash flight controllers (FC) over Bluetooth directly from the Configurator

### Describe the solution you'd like

Add a defined way to flash FC over Bluetooth in the Configurator

### Describe alternatives you've considered

To use the Speedybee app, but requires a speedybee FC and works from mobile app only.

### Other information

I have implemented BLE support in the `dronebridge` project in https://github.com/DroneBridge/ESP32/pull/103. So I was looking to add BLE flashing aswell to this, but the configurator has to support that.

haslinghuis, 2025-04-08:
Have been working on this - but obvious not ready nor working.
- transfer works - but my code does not erase or verify.
- not sure if the command for putting the FC into DFU is processed correctly.
- takes 30 minutes to transfer firmware but using gzip we could reduce this to about 10 minutes - but would need to be decompression on the host.

Alternatives
- Using wifi hotspot - perhaps using ELRS hardware.
- OTA functionality like https://docs.silabs.com/wisun/2.5.0/wisun-stack-api/sl-wisun-ota-dfu-api using coap

haslinghuis, 2025-04-08:
Have been working on this - but obvious not ready nor working.
- transfer works - but my code does not erase or verify.
- not sure if the command for putting the FC into DFU is processed correctly.
- takes 30 minutes to transfer firmware but using gzip we could reduce this to about 10 minutes - but would need  decompression support on the BLE host.

Alternatives
- Using wifi hotspot - perhaps using ELRS hardware.
- OTA functionality like https://docs.silabs.com/wisun/2.5.0/wisun-stack-api/sl-wisun-ota-dfu-api using coap

haslinghuis, 2025-04-08:
Have been working on this - but obvious not ready nor working.
- transfer works - but my code does not erase or verify.
- not sure if the command for putting the FC into DFU is processed correctly.
- takes 30 minutes to transfer firmware but using gzip we could reduce this to about 10 minutes - but would need  decompression support on the BLE host.

Alternatives
- Using wifi hotspot - perhaps using ELRS hardware (supports gzip)
- OTA functionality like https://docs.silabs.com/wisun/2.5.0/wisun-stack-api/sl-wisun-ota-dfu-api using coap

Witty-Wizard, 2025-04-09:
> Have been working on this - but obvious not ready nor working.
> - transfer works - but my code does not erase or verify.
> - not sure if the command for putting the FC into DFU is processed correctly. [...]

Where can I find the code for this, I would like to make my BLE stack compatible with this.

haslinghuis, 2025-04-09:
Did not know about DroneBridge - have used UAVmatrix with zerotier over LTE on RPi 3B+ with Navio2 flight controller using ArduPilot with QGC around 2018 but latency was around 100ms. Have not tried WBC hardware though.

Working to finish refactoring serial in https://github.com/betaflight/betaflight-configurator/pull/4413 but stuck on last commit and requires lot of testing all scenario's before I push the next one.

Using OTA or WiFi for flashing firmware is preferred.

Perhaps we can use ELRS hardware which already supports gzip but not sure ESP32 can hold such amount of data without pass-through using buffering. Otherwise would like to have a BF supported communications chip for new flight controllers.

Witty-Wizard, 2025-04-10:
What FC are you using for testing?

haslinghuis, 2025-04-10:
SPBE F405MINI or SPBE F722V3

Witty-Wizard, 2025-04-10:
> SPBE F405MINI or SPBE F722V3
> 
> Need to check HBRO KAKUTEH7

Hv they published how are they handling firmware updates through BLE, i could not find any where how they handled it