betaflight/betaflight-configurator #4411: Application not responding sonoma mac m2

binkybarnes, 2025-04-05:
### Describe the bug

I install latest version 10 dmg, and go to open it and it bounces for some time and then says not responding. 

[image]

### To Reproduce

try to open betaflight configurator on mac 

### Expected behavior

i expected it to open

### Configurator version

10.10.0

### Flight controller configuration

```txt

```

haslinghuis, 2025-04-05:
Try https://app.betaflight.com

binkybarnes, 2025-04-08:
ok it was supposed to give me the application is damaged error, but it didnt because i had a docker is damaged error, so the betaflight error wouldnt popup. i always just hid the docker popup in the side, but i fixed it now.