betaflight/betaflight-configurator #4417: 8kHz blackbox for perfect PID tune

boba575, 2025-04-10:
### Is your feature request related to a problem? Please describe

I'm using the black box with Matlab for PID tuning. This way I could get a really good PID tune.
I calculate the transfer function of my drone (what will the drone do for a given input), from the Setpoint and GyroADC values. This works very well for yaw axis however Matlab cannot simulate the system for pitch or roll axis-es, this is because the control authority is much higher there, so greater angular accelerations can be achieved. The 4kHz sampling rate is just too slow to record it.

### Describe the solution you'd like

Implement  8khz option for black box

### Describe alternatives you've considered

The only alternative for this would be to attach a separate micro controller with gyro to the drone for logging. This wouldn't be feasible for small drones.

### Other information

I cannot really calculate the exact frequency that would be needed (and it changes from drone to drone anyway), however I'm sure that 8khz would be enough, I searched for the maximum differences in following lines in the GyroADC columns, (I tried to estimate the angular acceleration with this). For Yaw it was 26 and for Pitch 36 and Roll 48. We know that the maximum value for this is between 26 and 36 (Yaw could be calculated Pitch couldn't). Since 48 is that 2*26 8kHz will be enough.

boba575, 2025-04-14:
Was it useing too much CPU, or why did it make it unflyable?

nerdCopter, 2025-04-14:
- resource hogging would make the drone rather "twitchy and unresponsive".
- if lots of logging data is turned off, then maybe it would work fine.

boba575, 2025-04-14:
Is there a way I could try it with only a few logged data? I only need the setpoint and the gyroADC for the pid tune

haslinghuis, 2025-04-14:
- We already support 1/1 (8Khz) logging ?
- Disable all fields you won't need

[image]

haslinghuis, 2025-04-14:
- We already support 1/1 (8KHz) logging ?
- Disable all fields you won't need

[image]

closed 2025-04-14 by haslinghuis