betaflight/betaflight-configurator #4422: Aggressive Browser caching JS causes outdated code

radio-mistress, 2025-04-13:
### Describe the bug

Betaflight Configurator in a browser, can end up with stale versions of the JavaScript due to aggressive caching behavior in Chrome and other modern browsers. This results in users running outdated or broken code even though the latest version of the app is deployed.

### To Reproduce

receive/cache several updates from https://app.betaflight.com/, buttons stop working and other odd behavior. 

### Expected behavior

The latest deployed JS should load fresh versions on app load, especially after updates, without requiring users to clear their browser cache. 

### Configurator version

4.6

### Flight controller configuration

```txt

```

### Add any other context about the problem that you think might be relevant here

Some quick fixes would like putting the date into the files (remove ability to cache)
<script type="module" src="/src/js/browserMain.js?v=${Date.now()}"></script>
Or pinning the versions at build, (only cache files within this git hash.
<script type="module" src="/assets/browserMain.2c7d10.js"></script>

nerdCopter, 2025-04-14:
The web app already does this after only a few seconds:
[image]

radio-mistress, 2025-04-14:
I was only able to resolve this by clearing all the site data.

nerdCopter, 2025-04-14:
understood. i have had Chrome do unexpected things and would solve with storage/cache cleaner extension(s) and/or `--disable-extensions`.  (currently using [The cleaner](https://chromewebstore.google.com/detail/the-cleaner-delete-cookie/ogfjgagnmkiigilnoiabkbbajinanlbn), not necessarily my fave, but mostly works.)

Maybe @haslinghuis and @McGiverGim  will consider your proposal or have some ideas.

PR #4425 (2025-04-21): Fix caching issue
- fixes #4422

This pull request includes a change to the `src/index.html` file. The change appends a timestamp query parameter to the script sources to ensure that the browser fetches the latest versions of the scripts, avoiding potential caching issues.

* [`src/index.html`](diffhunk://#diff-ce434ef5cc161194b450135a3586eb89357fc270e8aaa8e336db5a04e8d52fbeL6-R7): Added a timestamp query parameter to the script sources for `common.js` and `browserMain.js` to prevent caching issues.
closed 2025-04-21 by VitroidFPV