betaflight/betaflight-configurator #4434: Can't flash SPRacingH7EXTREME in DFU mode.

hydra, 2025-04-23:
### Describe the bug

It doesn't seem to recognise the FC when in SPRacingDFU bootloader mode.

Note: I've not used a recent version of BF for many months and many things have changed.

See the following screenshot, showing 10.10.0 (`c97deaf`), 11.0.0 (a062a9c4`) latest, 11.0.0 (e806299c) older and windows device manager.
Note  that in 10.10.0 the DFU device is shown, but in the 11.0.0 version no device is shown.  When clicking the `--I can't find my device--` in the dropdown menu, a modal dialog is shown but does not list the DFU device. 

[image]

### To Reproduce

1) Go to https://app.betaflight.com/# in a chrome browser, or run `yarn preview` on the command line and open it locally.
2) observe that no DFU device shows up in the port list.

### Expected behavior

DFU device found and usable.

### Configurator version

110.0.0 (a062a9c4)

### Flight controller configuration

```txt
N/A DFU related.
```

### Add any other context about the problem that you think might be relevant here

Chrome 135.0.70949.97 / Windows 11 23H2 x64

hydra, 2025-04-23:
Problem originally found on the H7EXTREME, also tested with H7NANO, H7ZERO, H7RF, H7EF, H7EVO and H7NEO.

haslinghuis, 2025-04-23:
Can't find USB device is for VCP mode.

hydra, 2025-04-23:
Also, tried updating chrome to 135.0.7049.115, updated local code to latest version on master `a062a9c4e32748f5607794c026d662403eb19c53` and have tried with/without "Show all serial devices" in options.

Feels like there's some bug in the DFU detection code.

Windows OS build: 22631.5189

PR #4438 (2025-04-26): Fix dfu permissions
- fixes: #4434
closed 2025-04-26 by haslinghuis