betaflight/betaflight-configurator #4435: SPRacingH7 4.3.2 builds missing ('Failed to load remote firmware')

hydra, 2025-04-23:
### Describe the bug

Builds are missing.

### To Reproduce

Select any of the SPRacingH7EXTREME, SPRacingH7NANO, SPRacingH7ZERO, SPRacingH7RF
Click 'Load firmware (Online)'
Observe failure message: 'Failed to load remote firmware'

### Expected behavior

All listed firmware can be downloaded.

### Configurator version

11.0 (`master`), 10.10.0 (`c97deaf`)

### Flight controller configuration

```txt
N/A
```

### Add any other context about the problem that you think might be relevant here

Here are the builds I was flashing for customers, built from the `4.3.2` tag at `60c9521da6072e6c891bd30fd772e91a790f0b53`
[betaflight_4.3.2_SPRACINGH7EXTREME_60c9521da6.zip](https://github.com/user-attachments/files/19865395/betaflight_4.3.2_SPRACINGH7EXTREME_60c9521da6.zip)
[betaflight_4.3.2_SPRACINGH7NANO_60c9521da6.zip](https://github.com/user-attachments/files/19865393/betaflight_4.3.2_SPRACINGH7NANO_60c9521da6.zip)
[betaflight_4.3.2_SPRACINGH7ZERO_60c9521da6.zip](https://github.com/user-attachments/files/19865394/betaflight_4.3.2_SPRACINGH7ZERO_60c9521da6.zip)

I don't have an H7RF 4.3.2 build available, it needs making or removing from the list of available builds.

nerdCopter, 2025-04-23:
confirmed 404.
[image]

@blckmn ?

blckmn, 2025-04-23:
@hydra corrected for the three where the hex has been provided. If you make a build for the remaining outstanding item I can load it for you, otherwise I can have it excluded from the list.

blckmn, 2025-04-24:
Noticed the SPRACINGH7RF is not available for 4.3.2. I have removed that version from the target.

closed 2025-04-24 by blckmn