betaflight/betaflight-configurator #4440: GD32 MCU Support

jpwu123456, 2025-04-26:
### Is your feature request related to a problem? Please describe

Now I'm attempting to port Betaflight to GD32F4XX MCUs and encountered three issues:  
①  Betaflight Configurator 10.10.0 seems not support Betaflight 4.6.0 firmware 
② GD32 USB VCP is not recognized by Configurator（It can be recognized by Serial Port Assistant）
③  GD32 DFU bootloader can be detected by configurator but fails to flash firmware ("Failed to open USB device")

### Describe the solution you'd like

For ①，Should I port a version of Betaflight 4.5.2 to use with Betaflight Configurator 10.10.0；Or can I solve it by modifying the configurator?
For ②，Or can I solve it by modifying the configurator 10.10.0 in my computer? 
For ③，Could give me some advice to solve it？

### Describe alternatives you've considered

For ③，I tried to use the DFU tools of GD32 and I can update the firmware with it. But I have no idea about how can I  use the configurator to update the firmware.

haslinghuis, 2025-04-26:
- Cannot find correct info to add GD VCP productId
- Flashing should work using https://app.betaflight.com
- In options tab enable

[image]

jpwu123456, 2025-04-26:
> * Cannot find correct info to add GD VCP productId
> * Flashing should work using https://app.betaflight.com
> * In options tab enable
> 
> [image]

Thanks for your reply.
1.  I tried to use the app.betaflight.com.  But unfortunately,  it's still failed after I click the **Flash Firmware**  button.

[image]

2.  For the GD  VCP, I tried to add the productId Info in the file of Betaflight Configurator\js\main.js (V 10.10.0), then restart the Betaflight Configurator in  my computer. And the GD VCP can be recognized now.

haslinghuis, 2025-04-26:
@jpwu123456 which OS (and version) are you running ?

If on linux need to add dev rules for DFU.

jpwu123456, 2025-04-26:
> [@jpwu123456](https://github.com/jpwu123456) which OS (and version) are you running ?
> 
> If on linux need to add dev rules for DFU.

It's win 10.  And when I use the GD32 tools, I can connect the GD DFU and update the firmware.

[image]

haslinghuis, 2025-04-26:
This should be correct ?

[{ vendorId: 10473, productId: 393 }, // GD32 DFU Bootloader](https://github.com/betaflight/betaflight-configurator/blob/ddfc76217d2a8c2923058f20fbaf5a53727622e9/src/js/protocols/devices.js#L61)

jpwu123456, 2025-04-27:
> This should be correct ?
> 
> [{ vendorId: 10473, productId: 393 }, // GD32 DFU Bootloader](https://github.com/betaflight/betaflight-configurator/blob/ddfc76217d2a8c2923058f20fbaf5a53727622e9/src/js/protocols/devices.js#L61)

There is good news. I modified the permissions of the GD DFU device in Ubuntu, and then used betaflight configurator to update the firmware, and it was successful.  Although some errors were reported in the log, the program update was successful。

[image]

But why it can't be successful in Windows, I still have no idea. Maybe I need to confirm whether it is a driver problem

And there is another problem. The on-chip Flash size of the GD MCU I have is 3M, but beftaflight configurator recognizes it as 1M. How can I correctly identify the on-chip Flash size?

haslinghuis, 2025-04-27:
- We don't have any GD32 targets - as this would be defined in target.h using `FLASH_PAGE_SIZE` macro
- For GD32 perhaps needs another WinUSB driver using Zadig
- Unrelated but STM32G47x does not use `FLASH_PAGE_SIZE` macro.

jpwu123456, 2025-04-27:
> * We don't have any GD32 targets - as this would be defined in target.h using `FLASH_PAGE_SIZE` macro
> * For GD32 perhaps needs another WinUSB driver using Zadig
> * Unrelated but STM32G47x does not use `FLASH_PAGE_SIZE` macro.

I will use Zadig to updated WinUSB driver and try again.
Thank you very much for your help!

jpwu123456, 2025-04-28:
> I will use Zadig to updated WinUSB driver and try again. Thank you very much for your help!

After I updated the WinUSB driver, it,s success in windows10.

closed 2025-04-28 by jpwu123456