betaflight/betaflight-configurator #4443: app.betaflight.com - linux - google chrome

popbangshwang, 2025-04-27:
### Describe the bug

Hi, I'm using arch linux, betaflight 5in1 running firmware 4.5.2
app.betaflight.com error when trying to connect;
`` This firmware version is not supported. Please upgrade to firmware that supports api version 1.44.0 or higher. Use CLI for backup before flashing. CLI backup/restore procedure is in the documentation. Alternatively download and use an old version of the configurator if you are not ready to upgrade.

Background:
Arch linux, tried both the flatpak and the aur version of google chrome
Selecting connect successfully detects my device:- Betaflight STM32F411 (ttyACM0) - Paired

This FC is working fine on my Mac. I was able to use app.betaflight.com to upgrade it from 4.5.0 to 4.5.2

This issue persists regardless of the FC.
The issue is not present when using Win11 on the same pc

### To Reproduce

Install Arch Linux > install google chrome (either flatpak or aur) > goto app.betaflight.com > attempt to connect to the FC

### Expected behavior

Should connect to the FC and allow me to configure it.

### Configurator version

11.0.0 (ddfc762)

### Flight controller configuration

```txt

```

haslinghuis, 2025-04-27:
- Suspect chrome is running in sandbox mode.
- Running same ubuntu version as we use for CI - had to install a different version - see #3960

Closing report as it is a platform / browser related issue.

closed 2025-04-27 by haslinghuis

popbangshwang, 2025-04-28:
Issue resolved by adding my user to the uucp group - https://wiki.archlinux.org/title/Users_and_groups#User_groups

Thanks reddit for the help - https://www.reddit.com/r/fpv/comments/1k9l0yw/anyone_running_arch_i_need_some_help/

haslinghuis, 2025-04-28:
[image]