betaflight/betaflight-configurator #4444: Add link to Setup tab for users to download the connected drone's Cloud Build firmware binary (corresponding Log and Configuration already supported)

DavidAnson, 2025-04-27:
### Is your feature request related to a problem? Please describe

Someone who purchases a RTF or BNF drone with Betaflight already installed may want to save a copy of the factory-installed firmware build before attempting to flash their own - in case anything goes wrong. Some manufacturers provide a link to their firmware build, but not all do. And in the case of buying a drone from someone else OR forgetting what you did 3 months ago, there's currently no good option.

### Describe the solution you'd like

For firmware builds produced by Cloud Build, the Betaflight Configurator already provides links to the corresponding build log and configuration: <https://github.com/betaflight/betaflight-configurator/blob/ddfc76217d2a8c2923058f20fbaf5a53727622e9/src/js/tabs/setup.js#L371-L383>. These links are created by calling the Betaflight Build API and passing the current image's build key to the `/json` and `/log` endpoints. It looks like that API also provides access to the firmware binary via the `/hex` endpoint: <https://build.betaflight.com/api/index.html>. I've verified this works for 2 different drones/manufacturers.

It would be nice to expose this link in the UI for the benefit of folks who don't grovel Swagger docs for fun. :) If you agree and nobody has time to do this and the Configurator can be developed in a GitHub Codespace, I could send a PR for this - it appears to be a relatively simple UI-only change that works the same as the existing two links (which are already suppressed for non-Cloud Build firmware).

### Describe alternatives you've considered

N/A

### Other information

N/A

nerdCopter, 2025-04-28:
[image]

hex not exposed, but 3rd party support-id service does: e.g. https://betaflight-support-explorer.netlify.app/00aa984e-2830-418e-b14b-120c9ce8c0bc
[image]

DavidAnson, 2025-04-28:
Thanks, @nerdCopter, I wasn't aware of that other site! To be clear, I am proposing adding the hex download functionality shown in your second picture to the main configurator app where you show in the first picture. I think it's important and easy enough to be official.

As I understand, the backup that's recommended during flashing is a console diff of the settings and not the firmware hex file that would be needed to fully revert.

One other interesting thing I found last night is that even when the vendor provides a hex file to download, it may not be what's actually flashed on the device you got from them. In particular, the link for the  BETAFPV Air64 Freestyle is five days older than what's on the actual drone I got new in the box from them.

DavidAnson, 2025-04-28:
Enhancing backup may be good, but I'd suggest there is value in being able to download the current hex file without having to flash a new image.

DavidAnson, 2025-04-28:
Yes, as I understand it, the link to a hex file based on the current build key will always be the CURRENT firmware image. It is not possible to go back in time, but presuming people get their device in a working state, saving THAT firmware before changing things  should give them a good baseline image if they accidentally mess anything up later. :)

DavidAnson, 2025-04-29:
I know we haven't agreed on anything, but I wanted to see what the development process was like. Here's a draft commit for the proposed change:

<https://github.com/DavidAnson/betaflight-configurator/compare/master...DavidAnson:betaflight-configurator:issue-4444>

The tests don’t seem to run from within GitHub Codespaces, but I expect they would run in GitHub Actions for a proper PR. I have tried this build locally and it behaves as expected.

DavidAnson, 2025-04-29:
Haha, okay, thanks, I'll do that this evening!

Something to think about when you review the PR is whether the order of the log, config, and hex buttons matters in the UI. I put the new hex button after the other two in code, but in the UI that puts it to the left of both, arguably in the most "prominent" spot. I could easily be convinced to put it in the far right and/or re-order the existing two buttons if that matters to anybody. Note that I alphabetized the localization entry, but did not do so for the new button definition OR placement (for consistency) and would be happy to do that as well.

haslinghuis, 2025-04-29:
Some thoughts:
- hex will only be available when cloud build. Local builds do not have that info.
- perhaps `Firmware` as title and button like options.
- only have to add i18n translation for English language.
- hide this line for local builds.

DavidAnson, 2025-04-29:
Thanks! Replies:

- I am aware of this, thanks.
- I understand this to be a results to use "Firmware" as the button text instead of "Hex". I'm not sure about the reference to Options as that currently has its own "section" and opens a dialog. Were you thinking Firmware should be separate as well? (A picture would help, but I'm on my phone right now.)
- I figured that out from other PRs. :)
- Hiding for local builds is implicit with the current implementation because Log and Config are only for Cloud Build. If I put Firmware into its own section, this would need a dedicated `if` check, I think.

haslinghuis, 2025-04-29:
Meant additional line

[image]

So after Options we get Firmware (new row)

DavidAnson, 2025-04-29:
Ok, I'll do that and add those colons, too. FWIW, not sure about the redundancy of "Options: [Options]", but it's very clear. :)

haslinghuis, 2025-05-02:
Resolved in #4450

closed 2025-05-02 by haslinghuis