betaflight/betaflight-configurator #4466: Status Indicator in "Modes" not working - every 2nd

notyourbiz, 2025-05-13:
### Describe the bug

In Modes (almost) every 2nd Mode is not indicated correctly even though it works on the Quad.

e.g. 
- "Flip over after Crash" works on the Drone, but not indicated as active in "Modes"
- "Pit Mode" works in VTX, but not indicated as active in "Modes"
- "Angle" works in Drone, but not indicated as active in "Modes" 

Screenshot is only for showing what i mean, i have tested every mode individually to exclude Interference

[image]

### To Reproduce

Update to 10.10.0 and set Modes ?

### Expected behavior

Working as expected... 

### Configurator version

10.10.0 (c97deaf)

### Flight controller configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.5.1 May 12 2025 / 07:20:57 (77d01ba3b) MSP API: 1.46
# config rev: 53e2e0a
```

closed 2025-05-13 by haslinghuis

notyourbiz, 2025-05-13:
Oh...?

thx mate, makes life a little easier... ;-) even though i prefer dark mode and (when i`m in the field) working offline ;-)

haslinghuis, 2025-05-13:
Online app can be installed as offline app too.