betaflight/betaflight-configurator #4474: Throttle Curve Preview Incorrect Logic: MID Point Not Reflected in Output

sanchuanhehe, 2025-05-16:
### Describe the bug

When setting Throttle MID = 0.3 and Throttle EXPO = 0.6, the Throttle Curve Preview does not correctly reflect the expected output. Specifically, the preview shows that when the throttle input is 0.3, the output is incorrectly displayed as 0.5, instead of the expected 0.3.

This suggests that the throttle preview curve is drawn based on a fixed midpoint (likely 0.5), ignoring the user-defined Throttle MID value.

[image]

[image]

### To Reproduce

Open Betaflight Configurator

Navigate to PID Tuning > Throttle

Set Throttle MID = 0.3

Set Throttle EXPO = 0.6

Observe the curve in the Preview panel

Note that at throttle input = 0.3, the output is incorrectly shown as 0.5

### Expected behavior

The throttle curve preview should correctly reflect the user-defined midpoint. When Throttle MID = X, the curve should pass through the point (X, X). In this case, with Throttle MID = 0.3, the preview curve should output 0.3 when the input is 0.3.

### Configurator version

2025-05-16 @12:44:25 -- OS: Linux  2025-05-16 @12:44:25 -- Configurator: 11.0.0 (16c3b2e)

### Flight controller configuration

```txt

```

### Add any other context about the problem that you think might be relevant here

Operating system:
web
Additional context:
This issue appears to be isolated to the visual preview and does not affect actual flight behavior or FC output.
It seems the drawing logic assumes a fixed midpoint at 0.5 when rendering the curve, rather than dynamically adjusting based on the configured Throttle MID.

sanchuanhehe, 2025-05-16:
**Additional context (continued):**
The issue may originate from the throttle curve drawing logic in the `redrawThrottleCurve` function:

🔗 [pid\_tuning.js:1573 – redrawThrottleCurve](https://github.com/betaflight/betaflight-configurator/blob/master/src/js/tabs/pid_tuning.js#L1573)

It’s possible that the `Throttle MID` value is not properly incorporated into the curve interpolation or transformation logic during the rendering process. The preview may still be assuming a default `mid = 0.5`, even when the user has explicitly set a different value.

We recommend reviewing how the `throttleExpo` and `mid` values are handled inside `redrawThrottleCurve`, especially when computing the actual y-values for the preview points.

sanchuanhehe, 2025-05-16:
**Additional context (continued):**
This issue may be related to recent changes introduced in **PR [[#4439](https://github.com/betaflight/betaflight-configurator/pull/4439)](https://github.com/betaflight/betaflight-configurator/pull/4439)** by **@[[marc.silas.frank@gmail.com](mailto:marc.silas.frank@gmail.com)](mailto:marc.silas.frank@gmail.com)**, submitted 6 days ago.

That PR modified the throttle curve preview logic, and it’s possible that the handling of the `Throttle MID` parameter was inadvertently broken or defaulted to a fixed value (e.g., 0.5) during rendering.

Please check the `redrawThrottleCurve` function, particularly how it processes the `throttleExpo` and `midThrottle` values when plotting the preview curve. It currently does not appear to pass the correct `mid` value into the throttle curve calculation, leading to incorrect visual output.

nerdCopter, 2025-05-16:
yep. same as: https://github.com/betaflight/betaflight-configurator/pull/4419#issuecomment-2881148769

haslinghuis, 2025-05-16:
@marc-frank this needs to be resolved.

Suggest using old English math for firmware below 4.6 as separate legacy function and new math in new function.

sanchuanhehe, 2025-05-17:
> [@marc-frank](https://github.com/marc-frank) this needs to be resolved.
> 
> Suggest using old English math for firmware below 4.6 as separate legacy function and new math in new function.

Did Betaflight 4.6 introduce a new mathematical function for the throttle curve?

marc-frank, 2025-05-18:
> > [@marc-frank](https://github.com/marc-frank) this needs to be resolved.
> > Suggest using old English math for firmware below 4.6 as separate legacy function and new math in new function.
> 
> Did Betaflight 4.6 introduce a new mathematical function for the throttle curve?

yes the hover point parameter is new

> [@marc-frank](https://github.com/marc-frank) this needs to be resolved.
> 
> Suggest using old English math for firmware below 4.6 as separate legacy function and new math in new function.

i just got around to looking at this again, sorry. i saw your pr. that looks good, too. better than wrapping the control point position calc in an if(semver)

nerdCopter, 2025-05-19:
another option is every time thr-mid adjusted, set thr-hover the same value. might be easier.

closed 2025-05-21 by haslinghuis