betaflight/betaflight-configurator #4493: mDNS support

haslinghuis, 2025-05-28:
### Is your feature request related to a problem? Please describe

PWA does not provide

### Describe the solution you'd like

Single code base for all platforms.

### Describe alternatives you've considered

NWjs

sugaarK, 2025-05-29:
I think for expediency we can put iOS to the side for now as it has horrible serial support any way. or though BLE and ELRS passthrough via wifi mitigate that.

haslinghuis, 2025-06-01:
### Proposal: Use Tauri 2.0 for Native Desktop Application

🎯 **Objective**

Transition from a purely browser-based deployment of the Betaflight Configurator to a native desktop application using Tauri 2.0, while preserving the current PWA frontend.

✅ **Benefits of Tauri 2.0**

- Lightweight native wrapper around web frontends (Rust backend)
- Secure and sandboxed communication
- Cross-platform (Windows, macOS, Linux)
- Easy integration with OS-level features (USB, networking, mDNS)

🧱 **Phase 1: Add Initial Tauri Support**

- [ ] Status - #4379

Tasks:

- [ ] Add Tauri 2.0 as optional backend for the project.
- [ ] Configure tauri.conf.json to point to Vite build output.
- [ ] Add basic Tauri build and dev scripts (pnpm tauri dev, pnpm tauri build).
- [ ] Ensure clean separation of web vs native contexts (via feature detection or platform guards).

🔌 **Phase 2: Add Serial Support**

- [ ] Plugin: [tauri-plugin-serial](https://lib.rs/crates/tauri-plugin-serialplugin)

Tasks:

- [ ] Install the plugin in Cargo.toml
- [ ] Register the plugin in src-tauri/src/main.rs
- [ ] Expose serial open/write/read functionality to frontend via JavaScript bindings
- [ ] Ensure compatibility with Betaflight USB devices (auto-detection, correct baud rates)

🌐 **Phase 3: Add Network Plugin**

- [ ] Plugin: [tauri-plugin-network](https://lib.rs/crates/tauri-plugin-network)

Use Cases:

- Detect current network state and interfaces (used for device auto-discovery and status UI)

Tasks:

- [ ] Add to Cargo.toml and register plugin
- [ ] Provide utility commands to frontend (e.g., get IP address, interface list)

🌍 **Phase 4: Add mDNS Support**

- [ ] Add library: [mdns-sd](https://github.com/keepsimple1/mdns-sd)

Use Case:

- Auto-discover flight controllers or telemetry devices over LAN (e.g., WiFi-enabled devices)

Tasks:

- [ ] Create Tauri command to asynchronously browse for _betaflight._udp.local, _http._tcp.local, etc.
- [ ] Return list of IP addresses + ports to frontend
- Optionally: implement mDNS advertisement for PC-hosted services

📦 **Packaging & Distribution**

- [ ] Package app via pnpm tauri build
- [ ] Generate platform-specific binaries (AppImage, MSI, DMG, etc.)
- [ ] Sign and notarize if targeting macOS and Windows

🔄 **Ongoing Considerations**

- [ ] Fallback to Web Serial when Tauri not available (pure browser use case)
- [ ] Use feature flags or tauri.isAvailable to toggle native functionality
- [ ] Consider eventual migration to full native runtime (e.g., Capacitor for mobile + Tauri for desktop)
- [ ] Requires changes to serial, serial_backend and port_handler
- [ ] Requires changes to STM32, DFU protocol and related MSP helpers.

📋 **Summary Table**

Feature | Plugin / Tool | Status
--- | --- | ---
Tauri 2.0 | [PR #4379](https://github.com/betaflight/betaflight-configurator/pull/4379) |	✅ In Progress
Serial Port | tauri-plugin-serial | ⏳ Planned
Network Info | tauri-plugin-network |⏳ Planned
mDNS Browser | mdns-sd crate |⏳ Planned

closed 2026-01-14 by haslinghuis