betaflight/betaflight-configurator #4502: Fedora 41: STM32 BOOTLOADER Failed to open serial port

vmorris, 2025-06-07:
### Describe the bug

I am running Fedora 41 on x86.
I tried to flash firmware on my MATEKSYS BetaFlight FC F405-OSD STM32F405RGT6 today both v10.10.0 portable and the v11 webapp, both in Chrome and Chromium (not in snap or flatpak).

I've tried adding all the normal udev rules and more, plus have added my user to all the suspect groups.

Still, Betaflight Configurator always returns "Failed to open serial port" whenever I try to communicate with the device.

Is there some more fundamental way I can diagnose what's wrong here? Are there some additional debugging messages I can get out of Betaflight Configurator?

```bash
$ getent group | grep vance
wheel:x:10:vance
dialout:x:18:vance
libvirt:x:983:vance
vance:x:1000:
```

```bash
$ cat /etc/udev/rules.d/45-stdfu-permissions.rules
# DFU (Internal bootloader for STM32 and AT32 MCUs)
'SUBSYSTEM=="usb", ATTRS{idVendor}=="2e3c", ATTRS{idProduct}=="df11", MODE="0664", TAG+="uaccess"'
'SUBSYSTEM=="tty", ATTRS{idVendor}=="2e3c", ATTRS{idProduct}=="df11", MODE="0664", TAG+="uaccess"'
'SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0664", TAG+="uaccess"'
'SUBSYSTEM=="tty", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0664", TAG+="uaccess"'
```

```bash
$ cat /etc/udev/rules.d/99-platformio-udev.rules
# Copyright (c) 2014-present PlatformIO <contact@platformio.org>
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#    http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

#####################################################################################
#
# INSTALLATION
#
# Please visit > https://docs.platformio.org/en/latest/core/installation/udev-rules.html
#
#####################################################################################

#
# Boards
#

# CP210X USB UART
ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea[67][013]", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"
ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="80a9", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"

# FT231XS USB UART
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6015", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"

# Prolific Technology, Inc. PL2303 Serial Port
ATTRS{idVendor}=="067b", ATTRS{idProduct}=="2303", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"

# QinHeng Electronics HL-340 USB-Serial adapter
ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"
# QinHeng Electronics CH343 USB-Serial adapter
ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="55d3", MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"
[... 259 lines cut ...]
ID_REVISION=2200
ID_USB_MODEL=STM32_BOOTLOADER
ID_USB_MODEL_ENC=STM32\x20\x20BOOTLOADER
ID_USB_MODEL_ID=df11
ID_USB_SERIAL=STMicroelectronics_STM32_BOOTLOADER_335F38563436
ID_USB_SERIAL_SHORT=335F38563436
ID_USB_VENDOR=STMicroelectronics
ID_USB_VENDOR_ENC=STMicroelectronics
ID_USB_VENDOR_ID=0483
ID_USB_REVISION=2200
ID_USB_INTERFACES=:fe0102:
ID_VENDOR_FROM_DATABASE=STMicroelectronics
ID_MODEL_FROM_DATABASE=STM Device in DFU Mode
ID_PATH_WITH_USB_REVISION=pci-0000:02:00.0-usbv2-0:4
ID_PATH=pci-0000:02:00.0-usb-0:4
ID_PATH_TAG=pci-0000_02_00_0-usb-0_4
ID_FOR_SEAT=usb-pci-0000_02_00_0-usb-0_4
ID_MM_DEVICE_IGNORE=1
ID_MM_PORT_IGNORE=1
MAJOR=189
MINOR=6
TAGS=:seat:
CURRENT_TAGS=:seat:

```

vmorris, 2025-06-07:
I tried a different flight controller and it seems to be working okay.
MTKS/MATEKF411RX(STM32F411)

Does this indicate my F405 is dead?

Thanks.

haslinghuis, 2025-06-07:
Connect flight controller while holding boot button. DFU should show.
Can further debug with `sudo dmesg` or using `dfu-util`